zephyrproject-rtos/zephyr #10200: net: sockets: Listen ignores backlog parameter

leodido99, 2018-09-25:
When calling listen on a socket you are supposed to provide the length of the pending connections backlog. In Zephyr this parameter is ignored when reaching net_context_listen.

Listen man entry specifies the behaviour as such:
The  backlog  argument defines the maximum length to which the queue of pending connections for sockfd may grow.  If a connection request arrives when  the  queue  is  full,  the  client  may receive  an  error with an indication of ECONNREFUSED or, if the underlying protocol supports retransmission, the request may be ignored so that a later reattempt at connection succeeds.

Currently the behaviour seems to be largely driven by the value of CONFIG_NET_MAX_CONN.

pfalcon, 2018-09-28:
@leodido99, ignoring backlog param, or handling it with "extended semantics" is a known pattern across TCP/IP stack implementations. Do you have a specific usecase behind this feature? Otherwise, this would be one of the lowest priorities feature request in the area of networking.

leodido99, 2018-10-01:
I don't really have any use case right now, was merely reporting this because it took me some time to find that I should use CONFIG_NET_MAX_CONN. I was more thinking of just having this written somewhere in the documentation maybe?

pfalcon, 2018-10-10:
@leodido99 : Just to reply, sounds good. The docs for sockets definitely needs improvements, and this fact can be mentioned too. But it's likely going to be backlogged (at least) until 1.14 closing to the release freeze. Because so far, there're a lot of code work to do...

rljordan, 2019-03-25:
Is this a bug or an enhancement?  Do we have an update on it?

closed 2025-09-18 by cfriedt