zephyrproject-rtos/zephyr #10207: Shell should accept either CR or LF as line delimiter

aescolar, 2018-09-25:
Currently the shell only accepts CR (\r) as the marker for end of line.
But different terminals/terminal configurations will insert a LF (canonical end of line), and some terminal driver configurations (the canonical one with ICRNL (default)) will replace CR with LF.
Moreover Unix terminals can also be configured with IGNCR (ignore CR), which means the character is dropped from the stream.

jarz-nordic, 2018-09-25:
Hi,
Currently shell accepts EOL marker defined by macro: 
`SHELL_DEFINE(uart_shell, "uart:~$ ", &shell_transport_uart, '\r', 10);` In this case `'\r'`

Is it requested that shell shall react on both at the same time?

aescolar, 2018-09-25:
> Is it requested that shell shall react on both at the same time?

**Either** of them would seem logical to me.

jarz-nordic, 2018-09-25:
Ok, I will rework that.
Please let me know if it is possible that some terminal will send both at the same time?
For example: `\r\n` or `\n\r` ?

aescolar, 2018-09-25:
> Shell needs to distinguish..

Does it? I mean, wouldn't the only result of not detecting it be an extra prompt displayed?

jarz-nordic, 2018-09-25:
Yes.
I would like to avoid extra empty prompt display.

PR #10328 (2018-10-02): Shell fix cr or lf as line delimiter
Shell macro `SHELL_DEFINE` no longer needs new_line character as an input parameter. Shell accepts both: `\r` and `\n` as a as line delimiter.

Fixes: #10207.
commit 794bda73 (2018-10-02): doc: shell: update documentation to reflect bug fix #10207

Shell macro SHELL_DEFINE no longer needs new line character.
Specification has been updated accordingly.
commit c471614c (2018-10-02): subsys: shell: fix accept either CR or LF as as line delimiter

1. Shell will accept CR or LF as line delimiter.
2. Macro SHELL_DEFINE simplified - it no longer requires
   new line character.
3. Fixes: #10207.
closed 2018-10-02 by nashif