zephyrproject-rtos/zephyr #10231: zephyr supported targets fail to flash with "ImportError: No module named 'colorama'"

pswarnak, 2018-09-26:
Tried flashing on multiple physical(non qemu) boards, all the boards fail to flash with the following message:
```
[0/1] Flashing nrf52_pca10040
Traceback (most recent call last):
  File "/home/pswarnak/workspace/clean_zephyr/zephyr/scripts/meta/west/main.py", line 12, in <module>
    import colorama
ImportError: No module named 'colorama'
FAILED: zephyr/cmake/flash/CMakeFiles/flash
cd /home/pswarnak/workspace/clean_zephyr/zephyr/samples/hello_world/build && /usr/bin/cmake -E env PYTHONPATH=/home/pswarnak/workspace/clean_zephyr/zephyr/scripts/meta /usr/bin/python3 /home/pswarnak/workspace/clean_zephyr/zephyr/scripts/meta/west/main.py flash --skip-rebuild
ninja: build stopped: subcommand failed.
```

**Zephyr commit:** 55b462cdfa8dcda620dc1cc984137ee1887e92a7.
**Toolchain:**  ZEPHYR_TOOLCHAIN_VARIANT=zephyr

**Steps to reproduce:**
cd zephyr/samples/hello_world
rm -rf build && mkdir build && cd build
cmake -D BOARD=nrf52_pca10040 -GNinja ../
ninja flash

pswarnak, 2018-09-26:
@mbolivar 
I see the following code lines added in scripts/meta/west/log.py
```
+import colorama
```

This seems to be causing build failure. Can you update the zephyr documentation - Development Environment Setup with the new dependency, if colorama package needs to be installed.
FYI: I am using Fedora Linux machine. Thank you.

mbolivar, 2018-09-26:
Thanks for reporting! Fixed in https://github.com/zephyrproject-rtos/zephyr/pull/10248. Sorry about that.