zephyrproject-rtos/zephyr #10240: sys_reboot for native posix missing

vsaw, 2018-09-26:
`sys_reboot` is currently not implemented for native posix.

See also https://github.com/zephyrproject-rtos/zephyr/issues/6044

vsaw, 2018-09-27:
Hi @aescolar, my application has [`CONFIG_REBOOT`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_REBOOT.html?highlight=reboot#cmdoption-arg-config-reboot) enabled and uses it on the NRF board. For the Posix Native target, the compilation fails because the Posix Arch does not implement the following function

```C
void sys_arch_reboot(int type);
void sys_clock_disable(void);
```

As a quick work-around I implemented mocks to make the code build. If there is interest I can upstream my mock implementation.

I have not looked into a "more useful" implementation of reboot on Posix.

aescolar, 2018-09-27:
> If there is interest I can upstream my ..

You are welcome to.

PR #10308 (2018-10-09): native_posix: Add support for CONFIG_REBOOT
##### Description of Change
To allow apps to compile with ```CONFIG_REBOOT```, add a weak stub for ```sys_arch_reboot()``` which does nothing in the POSIX arch. POSIX boards may choose to provide a more sensible implementation if relevant.
+
In the native_posix timer driver: Add support for ```CONFIG_SYSTEM_CLOCK_DISABLE```.

Fixes #10240

##### Checklist
- [x] PR description included and stakeholders cc'd
- [x] `sanitycheck` passes

##### Release Notes
Notes: native_posix: added support for CONFIG_REBOOT
closed 2018-10-09 by nashif