zephyrproject-rtos/zephyr #10242: in_addr/in6_addr covers s_addr/s6_addr with #defines which causes conflicts

rlubos, 2018-09-26:
I have question about Zephyr's `in_addr`/`in6_addr` implementation inside `net/net_ip.h`. It uses `#defines` to access it's fields, like this:

https://github.com/zephyrproject-rtos/zephyr/blob/a6de97b3a42389f7c052870b0730bb196cd72b58/include/net/net_ip.h#L86

The problem is that it covers fields like `s_addr`/`s6_addr` with it's own `#define`. In result, when I try to combine Zephyr headers with some external headers that redefine these structures and uses Posix nomenclature, I run into conflicts, as every occurence of `s_addr` is expanded with Zephyr's `#define`. Of course I can `#undef` these symbols, but that's not a very clear approach.

So my question is - do we need this kind of complexity? Could we simplify it to something like this:

```
/** IPv6 address structure */
struct in6_addr {
	union {
		u8_t		s6_addr[16];
		u16_t		s6_addr16[8]; /* In big endian */
		u32_t		s6_addr32[4]; /* In big endian */
	};
};

/** IPv4 address */
struct in_addr {
	union {
		u8_t		s_addr8[4];
		u16_t		s_addr16[2]; /* In big endian */
		u32_t		s_addr; /* In big endian */
	};
};
```

It still should allow to access fields in a simillar way, while retaining Posix compability without extra `#defines`. I can rework it and submit a patch if we agree it's fine to change this.

pfalcon, 2018-09-26:
From Linux' /usr/include/netinet/in.h :

~~~
/* IPv6 address */
struct in6_addr
  {
    union
      {
        uint8_t __u6_addr8[16];
        uint16_t __u6_addr16[8];
        uint32_t __u6_addr32[4];
      } __in6_u;
#define s6_addr                 __in6_u.__u6_addr8
#ifdef __USE_MISC
# define s6_addr16              __in6_u.__u6_addr16
# define s6_addr32              __in6_u.__u6_addr32
#endif
  };
~~~

So, we know where it comes from, and boils down to being shy to rely on anonymous union support (btw, there're no such tricks for ipv4 address in this current linux).

Whether we can/should change that is another question, but I don't see any problems with that.

jukkar, 2018-09-27:
@rlubos please send a PR that removes the defines, they are not very useful in first place.

rlubos, 2018-09-27:
@pfalcon Thanks for explaination, I wasn't even aware that's Linux who does it this way. Extra knowledge always appreciated!
@jukkar Just submited PR that removes the defines.

PR #10257 (2018-09-27): net: Remove s_addr/s6_addr defines from in_addr/in6_addr
Replace #defines for `s_addr`/`s6_addr` etc. in `in_addr`/`in6_addr` structures
within `net_ip.h` with fixed fileds. This prevents intrusive behaviour of
`net_ip.h`, that expands every occurence of `s_addr`/`s6_addr` with it's own
define, even in other, non-related structures.

Fixes #10242
closed 2018-09-27 by jukkar