zephyrproject-rtos/zephyr #10243: native_posix linking issues under Ubuntu 18.04

pfalcon, 2018-09-26:
Here're 2 facts:

1. I didn't build for BOARD=native_posix for a while.
2. Yesterday I upgraded to Ubuntu 18.04, which uses gcc 7.3.0 and other fancies of a modern linux distro, like PIE executables and what not.

Trying to build for BOARD=native_posix now I get:

~~~
[ 98%] Linking C executable zephyr.elf
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `../libapp.a(main.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(thread_entry.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(crc32_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(crc16_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(crc8_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(mempool.c.obj)' is incompatible with i386:x86-64 output
~~~

With VERBOSE=1 it's:

~~~
[ 98%] Linking C executable zephyr.elf
cd /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/hello_world/build/p/zephyr && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/zephyr_prebuilt.dir/link.txt --verbose=1
ccache /usr/bin/gcc    CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr.elf -T linker.cmd -Wl,-Map=/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/hello_world/build/p/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive ../libapp.a libzephyr.a arch/posix/core/libarch__posix__core.a soc/posix/inf_clock/libsoc__posix__inf_clock.a boards/boards/posix/native_posix/libboards__posix__native_posix.a -Wl,--no-whole-archive kernel/libkernel.a CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -L/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/hello_world/build/p/zephyr -Wl,--gc-sections -Wl,--build-id=none -ldl -pthread -lm 
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `../libapp.a(main.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/x86_64-linux-gnu-ld: i386 architecture of input file `libzephyr.a(thread_entry.c.obj)' is incompatible with i386:x86-64 output
~~~

It clearly looks like -m32 is missing from the link command line, and indeed, re-running that command in the appropriate dir with -m32 added leads to a successful link and working executable.

pfalcon, 2018-09-26:
> native_posix builds fine in Ubuntu 18.04 w gcc 7.3.

Maybe for you, but not for me, see above.

> You need to install the gcc-multilib & g++-multilib After, remember to clean before rebuilding.

Sure, I did all that.

aescolar, 2018-09-26:
This is clean virtual machine with the packages needed to compile zephyr + cmake + the zephyr sdk
``` bash
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic
alpi@alpiX1804v:~/aescolar-ot$ uname -a
Linux alpiX1804v 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
alpi@alpiX1804v:~/aescolar-ot$ mkdir build ; cd build
alpi@alpiX1804v:~/aescolar-ot$ source zephyr-env.sh 
alpi@alpiX1804v:~/aescolar-ot$ cd build/
alpi@alpiX1804v:~/aescolar-ot/build$ cmake ../samples/hello_world/ -GNinja -DBOARD=native_posix
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.5", minimum required is "3.4") 
-- Selected BOARD native_posix
Zephyr version: 1.13.99
Parsing Kconfig tree in /home/alpi/aescolar-ot/Kconfig
Loading /home/alpi/aescolar-ot/boards/posix/native_posix/native_posix_defconfig as base
Merging /home/alpi/aescolar-ot/samples/hello_world/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/alpi/.cache/zephyr
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- 
*******************************
*** NOTE TO KCONFIG AUTHORS ***
*******************************

The behavior of Kconfig 'default' properties in Zephyr has changed. The 
earliest default with a satisfied condition is now used, instead of the 
last one. This is standard Kconfig behavior.

See http://docs.zephyrproject.org/latest/porting/board_porting.html#old-zephyr-kconfig-behavior-for-defaults.

To get rid of this note, create a file called 'hide-defaults-note' in the 
Zephyr root directory. An empty file is fine.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alpi/aescolar-ot/build
alpi@alpiX1804v:~/aescolar-ot/build$ ninja
[1/69] Preparing syscall dependency handling

[69/69] Linking C executable zephyr/zephyr.elf
alpi@alpiX1804v:~/aescolar-ot/build$ zephyr/zephyr.exe 
***** Booting Zephyr OS v1.9.0-7536-g73c217802 *****
Hello World! native_posix
^C
Stopped at 0.690s
alpi@alpiX1804v:~/aescolar-ot/build$ git log | head
commit 73c21780272d1f7f4382c9213c463f3ddbca5716
```

aescolar, 2018-09-26:
If you have updated your installation, maybe the gcc you are running is not the default one from the ubuntu  distro? In that case you need to install the gcc-mulitlib for that gcc package. (say gcc-7-multilib)

pfalcon, 2018-09-26:
I indeed upgraded from Ubuntu 16.04. The gcc is:

~~~
pfalcon@x230:~$ which gcc
/usr/bin/gcc
pfalcon@x230:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-16ubuntu3' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --with-as=/usr/bin/x86_64-linux-gnu-as --with-ld=/usr/bin/x86_64-linux-gnu-ld --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3) 
~~~

multilib packages:

~~~
$ dpkg -l | grep multilib
ii  g++-4.8-multilib                                            4.8.5-4ubuntu8                              amd64        GNU C++ compiler (multilib support)
ii  g++-5-multilib                                              5.5.0-12ubuntu1                             amd64        GNU C++ compiler (multilib support)
ii  g++-7-multilib                                              7.3.0-16ubuntu3                             amd64        GNU C++ compiler (multilib support)
ii  g++-multilib                                                4:7.3.0-3ubuntu2                            amd64        GNU C++ compiler (multilib files)
ii  gcc-4.8-multilib                                            4.8.5-4ubuntu8                              amd64        GNU C compiler (multilib support)
ii  gcc-5-multilib                                              5.5.0-12ubuntu1                             amd64        GNU C compiler (multilib support)
ii  gcc-7-multilib                                              7.3.0-16ubuntu3                             amd64        GNU C compiler (multilib support)
ii  gcc-multilib                                                4:7.3.0-3ubuntu2                            amd64        GNU C compiler (multilib files)
~~~

So, everything looks ok. Thanks for confirming that it works for you in a clean VM and for other hints. As you can see, I unassigned this from you, will try to dig further myself (though I'd appreciate hints if I get stuck). My plan is to check how cmake passes -m32 to link stage and why it might be dropped. (Will work on this as time permits.)

aescolar, 2018-09-26:
And this is how the link line looks like in the example above:
```
[69/69] : && ccache /usr/bin/gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr.elf  -T zephyr/linker.cmd -Wl,-Map=/home/alpi/aescolar-ot/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/arch/posix/core/libarch__posix__core.a zephyr/soc/posix/inf_clock/libsoc__posix__inf_clock.a zephyr/boards/boards/posix/native_posix/libboards__posix__native_posix.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -L/home/alpi/aescolar-ot/build/zephyr -Wl,--gc-sections -Wl,--build-id=none -ldl -pthread -m32 -lm && cd /home/alpi/aescolar-ot/build/zephyr && /usr/bin/python3 /home/alpi/aescolar-ot/scripts/check_link_map.py zephyr.map && /usr/bin/objdump -S zephyr.elf > zephyr.lst && /usr/bin/readelf -e zephyr.elf > zephyr.stat && /usr/bin/cmake -E rename zephyr.elf zephyr.exe
```
Are you sure you have deleted the build directory and rebuilt?

aescolar, 2018-09-26:
Maybe check with `alternatives` which gcc you are actually running.

aescolar, 2018-09-27:
@pfalcon : Given that this issue seems to be related to your particular compilers setup, would you consider it reasonable to close it? (You can still re-open it later if you want to add something)

pfalcon, 2018-09-27:
> Given that this issue seems to be related to your particular compilers setup,

That's not my evaluation of this issue, mine would be "CMake build system doesn't have reliable enough ways to detect i386 vs x86_64 systems, leading to build errors". It's true that I'm the first to report it, but that doesn't mean nobody else is affected - it's just nobody else reported it yet (you can just wait for an echo thru the years).

As I mentioned, I unassigned you from this ticket, thanks for triaging it. Let me assign it to myself to avoid any confusion. If you wouldn't like to receive further notifications on this ticket, there's "Unsubscribe" button on the right of page. Thanks.

aescolar, 2018-10-01:
@nashif : This really seems to be an issue with Paul's multiple compilers installation/packaging issues after a distro upgrade, and not related to Zephyr per se.

paulsykes, 2018-10-05:
@pfalcon: Using Ubuntu 16.04.10 I have the same issue, -m32 appears to be missing from the link line. This was not an upgrade, it was a recent fresh install.

```
[69/69] Linking C executable zephyr/zephyr.elf
FAILED: : && ccache /usr/bin/gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr.elf  -T zephyr/linker.cmd -Wl,-Map=/home/paul/zephyr/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/arch/posix/core/libarch__posix__core.a zephyr/soc/posix/inf_clock/libsoc__posix__inf_clock.a zephyr/boards/posix/native_posix/libboards__posix__native_posix.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -L/home/paul/zephyr/build/zephyr -Wl,--gc-sections -Wl,--build-id=none -ldl -pthread -lm && cd /home/paul/zephyr/build/zephyr && /usr/bin/python3 /home/paul/zephyr/scripts/check_link_map.py zephyr.map && /usr/bin/objdump -S zephyr.elf > zephyr.lst && /usr/bin/readelf -e zephyr.elf > zephyr.stat && /usr/local/bin/cmake -E rename zephyr.elf zephyr.exe
/usr/bin/ld: Warning: size of symbol `zephyr_app_main' changed from 26 in libapp.a(main.c.obj) to 5 in zephyr/kernel/libkernel.a(init.c.obj)
/usr/bin/ld: i386 architecture of input file `zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `libapp.a(main.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `zephyr/libzephyr.a(thread_entry.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `zephyr/libzephyr.a(crc32_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `zephyr/libzephyr.a(crc16_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `zephyr/libzephyr.a(crc8_sw.c.obj)' is incompatible with i386:x86-64 output
/usr/bin/ld: i386 architecture of input file `zephyr/libzephyr.a(mempool.c.obj)' is incompatible with i386:x86-64 output
```

gcc is:
```
$ which gcc
/usr/bin/gcc
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.10' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 
```
multilib packages:
```
$ dpkg -l | grep multilib
ii  g++-5-multilib                              5.4.0-6ubuntu1~16.04.10                            amd64        GNU C++ compiler (multilib support)
ii  g++-multilib                                4:5.3.1-1ubuntu1                                   amd64        GNU C++ compiler (multilib files)
ii  gcc-5-multilib                              5.4.0-6ubuntu1~16.04.10                            amd64        GNU C compiler (multilib support)
ii  gcc-multilib                                4:5.3.1-1ubuntu1                                   amd64        GNU C compiler (multilib files)
```
Would be interesting to try a 16.04 clean vm to see if it is broken?

pfalcon, 2018-10-05:
@paulsykes : Cool, thanks for confirmation and additional info. Previous to upgrade, I was on 16.04 and everything was ok for me. (16.04 was itself upgraded from 14.04.)

It's still on my todo to debug this.

pfalcon, 2018-10-05:
From the console log above: `zephyr/samples/hello_world`.

aescolar, 2018-10-05:
The posix arch cmake file sets the "-m32"

About 16.04 I just tried, and it works fine in a clean machine. Here the exact steps I followed:
```
This image: 
http://ftp.uni-kl.de/pub/linux/ubuntu-dvd/xubuntu/releases/16.04/release/xubuntu-16.04.5-desktop-amd64.iso

#Installed on clean VM (Virtualbox 5.2.4, new vdi), selecting during install:
#    Download updates while installing
#    Erase disk and install
  
#On first boot installed VirtualBox guest aditions & reboot
  
#Followed steps from https://docs.zephyrproject.org/latest/getting_started/installation_linux.html
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install --no-install-recommends git cmake ninja-build gperf \
ccache doxygen dfu-util device-tree-compiler \
python3-ply python3-pip python3-setuptools python3-wheel xz-utils file \
make gcc-multilib autoconf automake libtool librsvg2-bin \
texlive-latex-base texlive-latex-extra latexmk texlive-fonts-recommended

git clone https://github.com/zephyrproject-rtos/zephyr.git

#continue with steps from guide
cd zephyr
pip3 install --user -r scripts/requirements.txt

#I skip installing the sdk just to save time
sudo mkdir -p /opt/zephyr-sdk/
sudo chown banana.banana /opt/zephyr-sdk/
cat <<EOF > /opt/zephyr-sdk/sdk_version
0.9.3
EOF

cat <<EOF > ~/.zephyrrc
export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk
EOF
  
mkdir $HOME/cmake && cd $HOME/cmake
wget https://cmake.org/files/v3.8/cmake-3.8.2-Linux-x86_64.sh
yes | sh cmake-3.8.2-Linux-x86_64.sh | cat
echo "export PATH=$PWD/cmake-3.8.2-Linux-x86_64/bin:\$PATH" >> $HOME/.zephyrrc

#Trying it out:

cd ../zephyr/
source zephyr-env.sh
mkdir build ; cd build
cmake ../samples/hello_world/ -GNinja -DBOARD=native_posix
ninja
zephyr/zephyr.exe 
#=> It works..
```

pfalcon, 2018-10-05:
> The posix arch cmake file sets the "-m32"

Yeah, and I thought maybe it's conditional on some checks, but it's not:

~~~
zephyr_ld_options(
  -ldl
  -pthread
  -m32
)
~~~

Now the fun thing: there were a bunch of toolchain upgrades in 18.04 this weekend, which I installed. Ant trying to build now for BOARD=native_posix, everything links well (I tried hello_world and samples/subsys/shell/shell_module).

I don't known what may be involved. Maybe cmake tries to smartass thru app's link flags and remove something on its whim? (Btw, @aescolar, how do we deal with building native_posix on 32-bit system?)

Anyway, now I'm keen to close this, @paulsykes, let us know if you plan to investigate your case further and keeping the ticket open is beneficial.

pfalcon, 2018-10-05:
Btw, I don't use -GNinja, i.e. use ol' good makefiles.

pfalcon, 2018-10-10:
Ok, let's close this for now, can reopen if stable reproduction cases are provided.

closed 2018-10-10 by pfalcon

jimschm, 2018-10-27:
Sigh, I hit this starting from scratch with 18.04 minimal setup and following instructions. There are some steps missing.

aescolar, 2018-10-28:
@jimschm : It is quite annoying that this is happening to some people. But it is very difficult to know what is the problem from the descriptions provided so far in this issue.
Which app are you trying to build? Which cmake version? which Ubuntu version?
Are you able to provide a reproducible set of steps where it will happens?

jimschm, 2018-10-29:
Yeah, my feedback isn't super actionable, except that it is another independent report. What I did is described well above. I installed 18.04 fresh from Ubuntu into a VirtualBox VM, then followed online instructions closely, including SDK installation. I put SDK export lines into .zephyrrc which was not explicitly instructed but fairly obvious. This brought me to the point where I could run cmake, but then compiler etc. was not installed. I installed a few more every day packages (build-essential, gcc-multilib, g++-multilib). Then I got here.

Unfortunately I don't have the time to troubleshoot. I switched to 16.04, did exact same steps, and it worked fine.

pfalcon, 2018-10-29:
@jimschm: I guess, going forward, we need to collect some objective data for the working/non-working environments. As an initial step, can start with providing output for the following commands:

~~~
cmake --version
dpkg -l gcc gcc-multilib binutils libc6 cmake
~~~

For me (working env):

~~~
pfalcon@x230:~$ cmake --version
cmake version 3.8.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
pfalcon@x230:~$ dpkg -l gcc gcc-multilib binutils libc6 cmake
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                        Version            Architecture       Description
+++-===========================-==================-==================-============================================================
ii  binutils                    2.30-21ubuntu1~18. amd64              GNU assembler, linker and binary utilities
un  cmake                       <none>             <none>             (no description available)
ii  gcc                         4:7.3.0-3ubuntu2.1 amd64              GNU C compiler
ii  gcc-multilib                4:7.3.0-3ubuntu2.1 amd64              GNU C compiler (multilib files)
ii  libc6:amd64                 2.27-3ubuntu1      amd64              GNU C Library: Shared libraries
ii  libc6:i386                  2.27-3ubuntu1      i386               GNU C Library: Shared libraries
~~~

(As can be seen, my cmake doesn't come from distro, it's actually a manual install from ubuntu 16.04 times, when distro version was too low for Zephyr's need. Need to try distro version one of these days, but not right now, as I need to get work done, not debug build tools hell too.)

aescolar, 2018-10-30:
>  but then compiler etc. was not installed. 

We are actually not telling people to install gcc or gcc-multilib in the getting started guide.. The multilib dependency is only mentioned in the native_posix board page. I will do a PR to update the getting started list of packages for Ubuntu. (Maybe this issue is the result of running cmake the first time without the needed things, then installing them, and then running it again with a corrupted cmake working directory (?))

pfalcon, 2018-10-30:
> and then running it again with a corrupted cmake working directory (?)

Definitely not for me. rm -rf ... is my frequent command during Zephyr development (I don't trust Zephyr's own clean).

jimschm, 2018-10-31:
I am well versed with cmake. That said it's good to mention in the instructions that tool paths are cached in the build directory.

PR #20818 (2019-11-21): posix arch: Use zephyr_link_libraries() to set -m32
For some reason, some users have been facing a bizarre issue
in which the -m32 option was not being passed to the linker
by cmake when building for the POSIX arch as a 32bit target,
even though the option was actually supported.

Instead of using zephyr_ld_options() which checks if an
option is supported and drops it otherwise, use
zephyr_link_libraries()

Fixes #10243 (hopefully, some of the reasons)