zephyrproject-rtos/zephyr #10271: Unexpected Kconfig warnings during documentation build

dbkinder, 2018-09-27:
Building the documentation (``make htmldocs``)  is showing some new and unexpected warning messages:

```
warning: the default selection CODE_ITCM (defined at boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) of <choice> (defined at boards/arm/mimxrt1050_evk/Kconfig:9) is not contained in the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_HYPERFLASH (defined at boards/arm/mimxrt1050_evk/Kconfig:16, boards/arm/mimxrt1060_evk/Kconfig:16) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1050_evk/Kconfig:19, boards/arm/mimxrt1060_evk/Kconfig:19) is defined with a prompt outside the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_HYPERFLASH (defined at boards/arm/mimxrt1050_evk/Kconfig:16, boards/arm/mimxrt1060_evk/Kconfig:16) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1050_evk/Kconfig:19, boards/arm/mimxrt1060_evk/Kconfig:19) is defined with a prompt outside the choice
```

dbkinder, 2018-09-27:
@carlescufi any thoughts on what's causing this?

ulfalizer, 2018-09-28:
*Dropping in from vacation*

The problem seems to be overlap between `boards/arm/mimxrt1050_evk/Kconfig` and `boards/arm/mimxrt1060_evk/Kconfig`.

When building applications, only the Kconfig files for the board selected with `cmake -D<BOARD>` are parsed. The Kconfig files have `osource "$(BOARD_DIR)/Kconfig"`, where the environment variable `BOARD_DIR` is set to the board name before Kconfig runs.

When building the documentation, all Kconfig files are parsed (`BOARD_DIR` is set to `boards/*/*`).

Maybe some stuff could be factored out to get rid of the warnings. `boards/arm/mimxrt1050_evk/Kconfig` and `boards/arm/mimxrt1060_evk/Kconfig` are almost identical.

There's a copy-paste error at the end of `boards/arm/mimxrt1060_evk/Kconfig` btw: `endif # BOARD_MIMXRT1050_EVK` should be `endif # BOARD_MIMXRT1060_EVK`.

dbkinder, 2019-01-02:
Ah, so this is another "conditional documentation" problem. Sigh. We have API documentation completeness issues too because of #ifdefs.  Our tools don't comprehend documenting an API or struct (for example) that changes based on a compile time #ifdef.

dbkinder, 2019-01-21:
Number of warnings has increased lately:
```
warning: the default selection CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) of <choice> (defined at boards/arm/mimxrt1020_evk/Kconfig:9) is not contained in the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
warning: the default selection CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) of <choice> (defined at boards/arm/mimxrt1050_evk/Kconfig:9) is not contained in the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_HYPERFLASH (defined at boards/arm/mimxrt1050_evk/Kconfig:16, boards/arm/mimxrt1060_evk/Kconfig:16) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_ITCM (defined at boards/arm/mimxrt1020_evk/Kconfig:13, boards/arm/mimxrt1050_evk/Kconfig:13, boards/arm/mimxrt1060_evk/Kconfig:13) is defined with a prompt outside the choice
warning: the choice symbol CODE_HYPERFLASH (defined at boards/arm/mimxrt1050_evk/Kconfig:16, boards/arm/mimxrt1060_evk/Kconfig:16) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
warning: the choice symbol CODE_QSPI (defined at boards/arm/mimxrt1020_evk/Kconfig:16, boards/arm/mimxrt1050_evk/Kconfig:20, boards/arm/mimxrt1060_evk/Kconfig:20) is defined with a prompt outside the choice
```

dbkinder, 2019-02-11:
@ulfalizer @MaureenHelm What started as 7 Kconfig warnings is now 45 when we build the documentation, all dealing with these mimxrt10XX_evk boards :(

closed 2019-02-13 by nashif