zephyrproject-rtos/zephyr #10272: doc build failing - no module named 'colorama'

dbkinder, 2018-09-27:
Documentation build (``make htmldocs``) is failing with the unexpected warning:
```
WARNING: autodoc: failed to import module 'west.runners.core' the following exception was raised:
No module named 'colorama'
```

Looks to be while processing the ``west`` APIs via Sphinx's autodoc extension.  Is ``colorama`` now required for doc generation?

dbkinder, 2018-09-27:
... and the answer is yes.  We need to have ``colorama`` package installed to build documentation now.  Installing it with ``pip3 install --user colorama`` and doc build succeeds.

closed 2018-09-27 by dbkinder

mbolivar, 2018-09-27:
> @mbolivar Is `colorama` now required for doc generation?

Yup, as you already saw, the new dependency was added to scripts/requirements.txt today.