zephyrproject-rtos/zephyr #10305: RFC: Add pin mask for gpio_port_xxx

qianfan-Zhao, 2018-09-29:
gpio_port_{configure, disable_callback, enable_callback, read, write} always set the whole pins in the group to a same mode. But sometimes not all the pins in a group has the same function. 

Example: If we want set PA1, PA3, PA5, PA7, PA8, PA9 as output mode, the others in PORTA as input mode, output HIGH to those output pin when initilizer, there doesn't has a easier way, **must** set them one by one.

The param **pin** for gpio_{configure, write, read ...} are unused if access operation is GPIO_ACCESS_BY_PORT. So I want expand this param for bitmask in GPIO_ACCESS_BY_PORT mode. So the example above could write in a easier way, like this:

```
#define OUTPUT_PIN_MASK    (BIT(1) | BIT(3) | BIT(5) | BIT(7) | BIT(8) | BIT(9))
gpio_port_configure(dev, OUTPUT_PIN_MASK, GPIO_DIR_OUT);
gpio_port_write(dev, OUTPUT_PIN_MASK, 0xffffffff);

...
```

b0661, 2018-10-07:
Hi @qianfan-Zhao, I am also experimenting with masked port access. What do you think about the following:

gpio.h:

```
...

/**
 * Access GPIO port pin by pin number.
 */
#define GPIO_ACCESS_BY_PIN 0

/**
 * Access GPIO port en bloc.
 */
#define GPIO_ACCESS_BY_PORT 1

/**
 * Access GPIO port pin(s) by port pins mask.
 */
#define GPIO_ACCESS_BY_PORT_MASK 2

...

/**
 * @brief Write a data value to the port.
 *
 * Write the output state of a port. Only write the state to the pins
 * that are enabled by the mask.
 *
 * The state of each pin is
 * represented by one bit in the value.  Pin 0 corresponds to the
 * least significant bit, pin 31 corresponds to the most significant
 * bit.  For ports with less that 32 physical pins the most significant
 * bits which do not correspond to a physical pin are ignored.
 *
 * @param port Pointer to the device structure for the driver instance.
 * @param mask Mask.
 * @param value Value to set on the port.
 * @return 0 if successful, negative errno code on failure.
 */
static inline int gpio_port_write_masked(struct device *port, u32_t mask, u32_t value)
{
	return gpio_write(port, GPIO_ACCESS_BY_PORT_MASK, mask, value);
}

...

```

There is just one inline function added that makes clear that this is a masked access. All current code that uses gpio_port_write() stays untouched. A driver that does not support GPIO_ACCESS_BY_PORT_MASK can return an error.

I think a masked access to for gpio_port_configure is not really necessary, but if it is necessary for synchronous reconfiguration it should be done the same way (aka. gpio_port_configure_masked(port, mask, flags) -> gpio_config(port, GPIO_ACCESS_BY_PORT_MASK, mask, flags)).

qianfan-Zhao, 2018-10-07:
@b0661 your opinion is good that can compatible the previous code, Which way is better between create new api gpio_port_write_masked and add param 'mask' to gpio_port_write in yours opinion? I prefer add mask param.
is it necessary for compatible the old versions?

b0661, 2018-10-07:
@qianfan-Zhao 

> Which way is better between create new api gpio_port_write_masked and add param 'mask' to gpio_port_write in yours opinion?

I prefer gpio_port_write_masked  as it makes clear what is done. We also have gpio_pin_write to write to a single pin. Having an extra function for a specialized write operation seems already in the API.

> is it necessary for compatible the old versions?

I don´t know. Using an extra inline function makes this PR an addition which is not intrusive; maybe this is the main argument Also why should I create a mask if I want to write to the whole port?

carlescufi, 2018-10-16:
@qianfan-Zhao I think your proposal is reasonable. Why don't you send a PR where you add the new mask and call, and implement it for one particular platform? For all the others you can return `-ENOTSUPP`

qianfan-Zhao, 2018-10-16:
@carlescufi sure, I am doing this pr

carlescufi, 2020-02-11:
This has been addressed in the GPIO rework

closed 2020-02-11 by carlescufi