zephyrproject-rtos/zephyr #10313: net: sockets: Packets are leaked on TCP abort connection

leodido99, 2018-10-01:
On certain conditions, when using sockets some packets are leaked.

I experienced this problem on a Zephyr TCP echo server. What happens is that client is sending packets and server is echoing, now the connection is physically interrupted. The client notice this and close the connection on its side, the server however since it is not sending any data leaves it open. When the physical connection is re-established, the client will reconnect, the server at this point will have two open connections. Then the client will send new messages, on my setup the server then sends it back to the two open connections. In my exemple I received 5 packets, so I send them back with 5 separate call to send. Now TCP will try to send these messages and after the number of retry expires for the first packet it will do an abort connection as expected and release the net context.

Socket zsock_received_cb will be called once for the first packet that is with a NULL packet and will then mark the socket as EOF. The thing is that there are still 4 packets in the TCP retry queue and they have ref count = 2 (once when the socket create the packet, then once for tcp queue retry), at this point what I notice is that during the clean up of the net context in net_tcp_release, I have 4 packets with refcount = 2 that are only decremented to 1 and so aren't freed.

If I summarize, basically enqueue several packets using socket send function. The first packet triggers the TCP abort connection because connection is closed. The packets in the queue after this packet are not properly freed because their refcount is 2 and net_tcp_release only unrefs once. Since socket recv callback is not called for all these packets but the first one that triggered the abort, they are leaked and cannot be recovered.

Hopefully my explanation is clear enough.

leodido99, 2018-10-10:
I've been investigating this more and actually I think I was mistaken. I was using a weird hardware setup and when the setup was updated the problem went away, so I'm closing this issue.

leodido99, 2018-10-15:
I'm actually re-opening this issue, because I've experienced it again. The problem is caused when several packets in the TCP retry list have refcount > 1, they are all leaked.

jukkar, 2019-02-21:
I wonder if this is still an issue as this one is quite old. Also lot of fixes has happened in network packet management. Closing this for now, can be reopened if needed.

closed 2019-02-21 by jukkar