zephyrproject-rtos/zephyr #10318: It is not documented what YAML bindings do

SebastianBoe, 2018-10-01:
In the device tree documentation it is only documented what yaml bindings could in theory do, it is not documented what they actually are doing today.

For a user it is much more interesting to know what they are used for than what we might use them for in the future. For instance it should be possible to read the device tree documentation and then be able to answer the question:

> Does the presence of "required" property in a binding enforce that the property is required?

Extract of DT documentation:

> A YAML description must be provided for every device node that is to be a source of information for the system. This YAML description can be used for more than one purpose. It can be used in conjunction with the device tree to generate include information. It can also be used to validate the device tree files themselves. A device tree file can successfully compile and still not contain the necessary pieces required to build the rest of the software. YAML provides a means to detect that issue.

The only user-relevant part of that paragraph is:

> A YAML description must be provided for every device node that is to be a source of information for the system.

nashif, 2019-02-11:
@galak @erwango Can you please document this if that was not done already?

erwango, 2019-02-15:
Working on it

PR #13429 (2019-02-19): doc/guide/dts: DTS documentation add-on
Update dts documentation.
Rework some parts to provide more clarity and elaborate
yaml bindings usage.

Fixes #10318
commit 8223cfb1 (2019-02-19): doc/guide/dts: DTS documentation add-on

Update dts documentation.
Rework some parts to provide more clarity and elaborate
yaml bindings usage.

Fixes #10318
closed 2019-02-19 by nashif