zephyrproject-rtos/zephyr #10329: SystemView overflow event

MarekPieta, 2018-10-02:
When we enable SystemView, it always records events data to buffer since system starts. It may often cause overflow event (especially for bigger systems). 

I think we could add additional option in KConfig to let users decide if they want to record events since system start or since host application connects. Thanks to this we wouldn't have to allocate big buffer to store all the data collected by SystemView since system start.
This could be implemented easily - for example function SEGGER_SYSVIEW_Start may be called conditionally based on boolean in KConfig.

nashif, 2018-12-08:
@MarekPieta do you want to propose a patch for this?

nashif, 2019-01-30:
fixed in #12149

closed 2019-01-30 by nashif