zephyrproject-rtos/zephyr #10330: drivers: flash: API for check whether flash is empty "in place"

nvlsianpu, 2018-10-02:
For now it is required to read flash, an then to check whether whole bytes read were equal to the erased-value. This implies many calls to the flash API, more call is required if temporary read buffer is smaller. Checking whether flash range is empty seams to be a very common feature.
Introduction of a such will decrease execution and might reduce ROM footprint as well.

API above should have immediate API in flash_map module.

For 1s step I propose to implement a generic implementation using current internal flash interface (flash_api_read) which will be used for all vendors who won't implement dedicated callback (so initially for all).

nvlsianpu, 2018-10-02:
cc @lemrey @pizi-nordic @carlescufi

d3zd3z, 2019-12-02:
This would need to be optional, however, as not all flash devices can be directly read.