zephyrproject-rtos/zephyr #10336: Misra C doesn't require checking boolean return value against an integer in if statement

yashi, 2018-10-02:
The commit 92ea2f91892e83d401598985d2b5ba60a659fdfa adds "== 0" to check against functions returning an integer as well as a boolean.

Z_SYSCALL_MEMORY_ARRAY uses __builtin_umul_overflow() and this function returns a boolean.

Misra 2012 Rule 14.4 says if statement shall have boolean type thus comparing return value against integer make sense but checking boolean return value against zero doesn't seem to be right.

ceolin, 2018-10-02:
That's true this function returns a boolean and the check is wrong. But there is other problem here, this is a builtin function and it is not allowed too and since there is no prototype for this function they static analyzer interpret it as returning an integer.

Thanks for pointing it out.

ceolin, 2018-10-02:
#10342

ceolin, 2018-10-04:
integrated, thanks again.

closed 2018-10-04 by ceolin