zephyrproject-rtos/zephyr #10338: Add PyLint checking of all python scripts in CI

dleach02, 2018-10-02:
PyLint is an effective tool for static analysis of python code. Currently, some usage paths of the python scripts are not covered by CI and thus will miss errors.  Adding PyLint analysis of the python code will help ensure errors are caught at CI instead of after merge.

dleach02, 2018-10-03:
I manually ran PyLint3 on the various scripts and here is the current state:

************* Module dts.extract_dts_includes
dts/extract_dts_includes.py:100:24: E1101: Instance of 'dict' has no 'iteritems' member (no-member)
************* Module dts.devicetree
dts/devicetree.py:215:40: E0602: Undefined variable 'name' (undefined-variable)
************* Module dts.extract.directive
dts/extract/directive.py:51:4: E0211: Method has no argument (no-method-argument)
************* Module elf_helper
elf_helper.py:96:61: E1101: Instance of 'ArrayType' has no 'num_members' member (no-member)
elf_helper.py:153:16: E1101: Instance of 'AggregateTypeMember' has no 'debug_die' member (no-member)
elf_helper.py:516:24: E0203: Access to member 'sym' before its definition line 517 (access-member-before-definition)
elf_helper.py:516:39: E0203: Access to member 'sym' before its definition line 517 (access-member-before-definition)
************* Module gen_app_partitions
gen_app_partitions.py:47:12: E1101: Module 'os' has no 'exit' member; maybe '_exit'? (no-member)
************* Module gen_idt
gen_idt.py:108:8: E1121: Too many positional arguments for function call (too-many-function-args)

dleach@dleach-VirtualBox:~/work/zephyr/scripts/kconfig$ pylint3 * -E
************* Module checkconfig
checkconfig.py:147:7: E1101: Module 'kconfiglib' has no 'Config' member (no-member)
************* Module kconfiglib
kconfiglib.py:3147:20: E1101: Instance of 'dict' has no 'viewvalues' member (no-member)
************* Module menuconfig
menuconfig.py:836:29: E0601: Using variable '_show_name' before assignment (used-before-assignment)
menuconfig.py:2836:8: E0632: Possible unbalanced tuple unpacking with sequence: left side has 3 label(s), right side has 0 value(s) (unbalanced-tuple-unpacking)

dleach@dleach-VirtualBox:~/work/zephyr/scripts/meta$ pylint3 * -E
************* Module runners.intel_s1000
west/runners/intel_s1000.py:119:18: E1120: No value for argument 'cmd' in constructor call (no-value-for-parameter)

************* Module commands.run_common
west/commands/run_common.py:312:8: E1101: Module 'log' has no 'warn' member; maybe 'wrn'? (no-member)

************* Module sanitycheck
sanitycheck:1478:20: E1136: Value 'main_c' is unsubscriptable (unsubscriptable-object)
sanitycheck:1486:20: E1136: Value 'main_c' is unsubscriptable (unsubscriptable-object)
sanitycheck:1500:16: E1121: Too many positional arguments for function call (too-many-function-args)

************* Module sanity_chk.ini2yaml
sanity_chk/ini2yaml.py:3:0: E0401: Unable to import 'ConfigParser' (import-error)

ulfalizer, 2018-10-03:
Re. the ones I know stuff about:

 ```
************* Module checkconfig
checkconfig.py:147:7: E1101: Module 'kconfiglib' has no 'Config' member (no-member)
```

`checkconfig.py` hasn't been updated for Kconfiglib 2, which renamed `Config` to `Kconfig`.

```
************* Module kconfiglib
kconfiglib.py:3147:20: E1101: Instance of 'dict' has no 'viewvalues' member (no-member)
```

`pylint3` getting confused by Python 2/3 compatibility.

```
************* Module menuconfig
menuconfig.py:836:29: E0601: Using variable '_show_name' before assignment (used-before-assignment)
```

False positive due to global variable. I'm a fan of them under the right circumstances.

```
menuconfig.py:2836:8: E0632: Possible unbalanced tuple unpacking with sequence: left side has 3 label(s), right side has 0 value(s) (unbalanced-tuple-unpacking)
```

False positive. The code is `y, x, s = args[:3]`, and `args` is required to have at least 3 elements in that path.

dleach02, 2018-10-03:
I didn't inspect all of them but there are some that are real errors on that list... my only point is that integrating PyLint into the CI process will help catch all of these and any new ones... and it is just generally good practice with Python code (particularly if we open it up a bit and look at the warnings as well).

ulfalizer, 2018-10-03:
Yep, doesn't hurt to run it from time to check for errors.

I don't know if we've planned to take it further than that. I think there's a risk that it turns into tool pleasing, for little or negative benefit.

This guy is my spirit animal when it comes to "PEP 8-ifying" code and the like: https://www.youtube.com/watch?v=wf-BqAjZb8M

ulfalizer, 2018-10-04:
Just to not sound overly negative, I think it'd be awesome to automatically flag stuff like unused local variables. Getting rid of things like that is always a win.

dleach02, 2018-10-04:
I don't follow everything on PyLint/PEP. But for actual error checking it is very useful. I personally like Google's style: https://github.com/google/styleguide/blob/gh-pages/pyguide.md

ulfalizer, 2019-09-20:
This has been added now.

closed 2019-09-20 by ulfalizer