zephyrproject-rtos/zephyr #10345: The OpenAMP sample does not build

SebastianBoe, 2018-10-03:
The OpenAMP sample does not build on the latest master.

Steps to reproduce;

1. git checkout b941561e # Check out a recent master
2. $ZEPHYR_BASE/scripts/sanitycheck -T $ZEPHYR_BASE/samples/subsys/ipc/openamp

```
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD lpcxpresso54114_m4
Zephyr version: 1.13.99
Parsing Kconfig tree in /home/sebo/zephyr/Kconfig
Loading /home/sebo/zephyr/boards/arm/lpcxpresso54114_m4/lpcxpresso54114_m4_defconfig as base
Merging /home/sebo/zephyr/samples/subsys/ipc/openamp/prj.conf
-- Generating zephyr/include/generated/generated_dts_board.h
extract_dts_includes.py: Merge of 'category': 'required'  overwrites 'optional'.
-- Cache files will be written to: /home/sebo/.cache/zephyr
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sebo/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/arm-zephyr-eabi/arm-zephyr-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- C_FLAGS :  -Wall -Wextra
-- Looking for include file stdatomic.h
-- Looking for include file stdatomic.h - found
-- Looking for include file fcntl.h
-- Looking for include file fcntl.h - not found
-- Build type:  Debug
-- Host:    Linux/x86_64
-- Target:  Generic/arm
-- Machine: cortexm
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test
make[1]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[2]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
Scanning dependencies of target kobj_types_h_target
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
Scanning dependencies of target openamp_remote
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
[  1%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
Scanning dependencies of target syscall_macros_h_target
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
Scanning dependencies of target driver_validation_h_target
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
Scanning dependencies of target syscall_list_h_target
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
[  1%] Creating directories for 'openamp_remote'
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
[  1%] Generating include/generated/syscall_macros.h
make[3]: Entering directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
[  2%] Generating include/generated/driver-validation.h
[  2%] Generating misc/generated/syscalls.json
[  3%] No download step for 'openamp_remote'
[  3%] No patch step for 'openamp_remote'
[  4%] No update step for 'openamp_remote'
[  5%] Performing configure step for 'openamp_remote'
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
[... 259 lines cut ...]
-- Target:  Generic/arm
-- Machine: cortexm
CMake Error at ../../../../../CMakeLists.txt:537 (message):
  The Zephyr library 'drivers__ipm' was created without source files.  Empty
  (non-imported) libraries are not supported.  Either make sure that the
  library has the sources it should have, or make sure it is not created when
  it has no source files.

-- Configuring incomplete, errors occurred!
See also "/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test/openamp_remote-prefix/src/openamp_remote-build/CMakeFiles/CMakeOutput.log".
See also "/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test/openamp_remote-prefix/src/openamp_remote-build/CMakeFiles/CMakeError.log".
CMakeFiles/openamp_remote.dir/build.make:105: recipe for target 'openamp_remote-prefix/src/openamp_remote-stamp/openamp_remote-configure' failed
make[3]: *** [openamp_remote-prefix/src/openamp_remote-stamp/openamp_remote-configure] Error 1
make[3]: Target 'CMakeFiles/openamp_remote.dir/build' not remade because of errors.
make[3]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
CMakeFiles/Makefile2:168: recipe for target 'CMakeFiles/openamp_remote.dir/all' failed
make[2]: *** [CMakeFiles/openamp_remote.dir/all] Error 2
make[2]: Target 'all' not remade because of errors.
make[2]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'
Makefile:129: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Target 'default_target' not remade because of errors.
make[1]: Leaving directory '/home/sebo/zephyr/sanity-out/lpcxpresso54114_m4/samples/subsys/ipc/openamp/test'

```

SebastianBoe, 2018-10-03:
I can no longer reproduce this. I can't explain why.

SebastianBoe, 2018-10-03:
Found the explanation.

When openamp does a recursive-build (&¤#!) it does not correctly transfer all necessary metadata from one build to the other, specifically, the BOARD is set differently in the two builds. This can be seen if one greps for BOARD in above attached log output.

You will see that one build is done with nrf52840_pca10056 and the other with xpresso.

galak, 2018-10-28:
This change seems to pass the wrong board to the 'remote' build.

galak, 2018-10-28:
> You will see that one build is done with nrf52840_pca10056 and the other with xpresso.

Where do we think nrf52840_pca10056 was coming from?

galak, 2018-10-29:
> @galak : Steps to reproduce?

Do a build of open amp sample and look at the .config in the remote build dir, it will be for the M4 core instead of M0

SebastianBoe, 2018-10-29:
> Where do we think nrf52840_pca10056 was coming from?

It was coming from the environment.

galak, 2018-10-29:
> Do a build of open amp sample and look at the .config in the remote build dir, it will be for the M4 core instead of M0

`build/openamp_remote-prefix/src/openamp_remote-build/zephyr/.config` will show the issue and which SoC/board is getting set.

SebastianBoe, 2018-10-29:
I see that the sample documents that it only support the xpresso board. So I am going to solve this by erroring out when the user tries to specify a non-xpresso board.

PR #10905 (2018-10-29): cmake: openamp: Fix OpenAMP sample
Stop specifying that the OpenAMP 'remote' app should use the same
board as the root app.

Also, add assertions to make sure that the user does not try to
override the board's that are specified in the app build scripts.

This fixes #10345
commit b3d77a0c (2018-10-29): cmake: openamp: Fix OpenAMP sample

Stop specifying that the OpenAMP 'remote' app should use the same
board as the root app.

Also, add assertions to make sure that the user does not try to
override the board's that are specified in the app build scripts.

This fixes #10345
closed 2018-10-29 by nashif