zephyrproject-rtos/zephyr #10348: valgrind detected issue in logger

aescolar, 2018-10-03:
There seems to be an issue in the logger when it frees messages that `valgrind` detects:

==18932== Conditional jump or move depends on uninitialised value(s)
==18932==    at 0x804AC96: log_is_strdup (log_core.c:539)
==18932==    by 0x804AEEF: msg_free (log_msg.c:51)
==18932==    by 0x804AFB8: log_msg_put (log_msg.c:86)
==18932==    by 0x804A971: msg_process (log_core.c:369)
==18932==    by 0x804A9D2: log_process (log_core.c:383)
==18932==    by 0x804A4A3: msg_finalize (log_core.c:65)
==18932==    by 0x804A549: log_n (log_core.c:137)
==18932==    by 0x804A6AB: log_generic (log_core.c:193)
==18932==    by 0x8048F83: log_handler (ext_log_system_adapter.c:36)
==18932==    by 0x8048E6F: ext_log_system_foo (ext_log_system.c:17)
==18932==    by 0x80494FA: external_log_system_showcase (main.c:218)
==18932==    by 0x80495D7: log_demo_thread (main.c:267)

Steps to reproduce:
```
cmake ../samples/subsys/logging/logger/  -GNinja -DBOARD=native_posix  -DCONFIG_NO_OPTIMIZATIONS=y
ninja
valgrind zephyr/zephyr.exe
```

carlescufi, 2018-10-03:
@jarz-nordic if @nordic-krch is away could you take a look at this one?

jarz-nordic, 2018-10-03:
@carlescufi : We need to wait for Krzysztof, he shall be tomorrow at the office.

PR #10425 (2018-10-31): logging: add arguments counting in log_generic
Log_generic() can be used by external logger systems to map to zephyr
logger subsystem. Function has variable number of arguments. So far
logger assumed maximal number of arguments to avoid scanning
string to count number of arguments. In outcome, uninitialized memory
was treated as valid argument in a message (but discarded when actual
string formatting took place). Valgrind complained about that so
log_generic() has been extended to scan the string and count number
of arguments and build message with valid number of arguments.

Fixes #10348.
closed 2018-10-31 by carlescufi