zephyrproject-rtos/zephyr #10389: Conversion of net core to new logger breaks WiFi driver builds

GAnthony, 2018-10-05:
commit a76814bfb698a33bb6a5337421c1504b157beb71
" net: Convert core IP stack to use log levels"

seems to have broken the builds of WiFi drivers for both SimpleLink and winc1500.

I tried to grok what changes would be needed to update wifi_mgmt, or the drivers, but think I need some pointers.

Eg: try to build samples/net/drivers/wifi

`build[(no branch, bisect started on 3d1a659)]> pwd
/home/gpitney/zephyr/samples/net/wifi/build

build[(no branch, bisect started on 3d1a659)]> cmake .. -DBOARD=quark_se_c1000_devboard
-- Found PythonInterp: /usr/bin/python3.4 (found suitable version "3.4.2", minimum required is "3.4") 
-- Selected BOARD quark_se_c1000_devboard
Zephyr version: 1.13.99
Parsing Kconfig tree in /home/gpitney/zephyr/samples/net/wifi/Kconfig
Loading /home/gpitney/zephyr/boards/x86/quark_se_c1000_devboard/quark_se_c1000_devboard_defconfig as base
Merging prj_base.conf
Merging /home/gpitney/zephyr/samples/net/wifi/boards/quark_se_c1000_devboard.conf

warning: SYS_LOG_WIFI_LEVEL (defined at drivers/wifi/Kconfig:14) was assigned the value '1' but got
the value ''. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SYS_LOG_WIFI_LEVEL.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.

warning: SYS_LOG_SHOW_COLOR (defined at subsys/logging/Kconfig:23) was assigned the value 'y' but
got the value 'n'. You can check symbol information (including dependencies) in the 'menuconfig'
interface (see the Application Development Primer section of the manual), or in the Kconfig
reference at http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SYS_LOG_SHOW_COLOR.html
(which is updated regularly from the master branch). See the 'Setting configuration values' section
of the Board Porting Guide as well.

prj_base.conf:19: warning: attempt to assign the value "1" to the undefined symbol SYS_LOG_NET_LEVEL

prj_base.conf:25: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_CORE

prj_base.conf:26: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_CONTEXT

prj_base.conf:27: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_NET_PKT

prj_base.conf:28: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_UTILS

prj_base.conf:29: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_IF

prj_base.conf:30: warning: attempt to assign the value "n" to the undefined symbol NET_DEBUG_CONN

Error: Aborting due to non-whitelisted Kconfig warning 'prj_base.conf:19: warning: attempt to assign
the value "1" to the undefined symbol SYS_LOG_NET_LEVEL'. Note: If this warning doesn't point to an
actual problem, you can add it to the whitelist at the top of
/home/gpitney/zephyr/scripts/kconfig/kconfig.py.`

jukkar, 2018-10-05:
This is trivial to fix, I am working on it. Unfortunately this wifi sample was missing sample.yaml file so it was never checked by sanitychecker.

closed 2018-10-05 by nashif

GAnthony, 2018-10-05:
Thanks @jukkar.  It wasn't quite obvious to me the conversion needed.  Saw wifi_mgmt.c was still using SYS_LOG, so guess that one just wasn't completely converted yet.  I verified wifi driver builds and runs now.  
Thanks again!

commit d056d63e (2018-10-05): samples: net: wifi: Adding sample.yaml file

As the sample.yaml file was missing, then sanitychecker did
not testing any wifi drivers or other wifi related APIs.

Fixes #10389