zephyrproject-rtos/zephyr #10404: Clarification of the code style to use for module* params in Kconfig templates

pfalcon, 2018-10-05:
Recently, a "Kconfig template feature" was introduced, which requires assigning values to some template variables and then source a "Kconfig template" which will have those vars substituted.

This ticket is to clarify punctuation style to use of these template var assignments. E.g., initially syntax with spaces around `=`, e.g. https://github.com/zephyrproject-rtos/zephyr/commit/6aed72e487b5c3a8ccb5ecadab558567e16fee9c#diff-c329d91124f630eff1649946f71669b8R44 :

~~~
module = SHELL
module-str = Shell
~~~

However, PR https://github.com/zephyrproject-rtos/zephyr/pull/8816 
landed a bunch (17 hits in my grep) of using it without spaces (e.g. https://github.com/zephyrproject-rtos/zephyr/pull/8816/files#r221893651):

~~~
module-dep=LOG
module-str=Log level for Ethernet driver
module-help=Sets log level for Ethernet Device Drivers.
~~~

So, this ticket is to see if we care and calling to @ulfalizer as the author of templating feature to explain how it's intended to be.

P.S. This may seem minor, but nowaydays we do as insane things as stuffing "(void)" in front of each memset, so surely we can do a sane thing of notwritingstuffwithoutspaces? Also, all big mess starts from small cases like this.

pfalcon, 2018-10-05:
cc @jukkar

ulfalizer, 2018-10-05:
Wee... a soapbox. ;)

I like spaces around `=`. Looks less cramped, and avoids tricking people into thinking that whitespace is significant. Preprocessor variables are based on Make variables, and people usually put spaces around `=` and `:=` there too.

Another thing is that I'd use a plain Kconfig `if` instead of `module-dep`:

```
if LOG

module = SLIP
module-str = Log level for slip driver
source "subsys/net/Kconfig.template.log_config.net"

endif # LOG
```

It's a few more lines (the blank lines are negotiable), but makes it obvious at a glance that the `LOG` dependency applies to everything in the template.

In general, I think it's best to avoid using the preprocessor for things that can be handled as easily in plain Kconfig. It just adds another layer for people to untangle.

Being able to define templates like this is more a side effect of the new [Kconfig preprocessor](https://github.com/torvalds/linux/blob/master/Documentation/kbuild/kconfig-macro-language.txt) by the way. It could easily be abused to create harder-to-read Kconfig files, so we ought to be careful with it.

Oh... and if you need someone sympathetic to whine to about MISRA-related cargo-culting, I'm your guy.~

pfalcon, 2018-10-07:
@ulfalizer : Thanks for the detailed reply. Glad that you confirm preferences for spaces around "=", and since posting this ticket, the case which prompted it was resolved: https://github.com/zephyrproject-rtos/zephyr/pull/10406 , i.e. it wasn't the case that there're disagreement, it was just an omission in an otherwise pretty big and backlogged patch.

> Another thing is that I'd use a plain Kconfig if instead of module-dep:

> `if LOG`

Hmm, with me actually paying attention to this for the first time, I'd say I personally prefer module-dep - it's pretty useful feature to be able to abstract "control flow" to a template, this kinda makes it more "declarative" than "imperative", and easier to parse/reason about.

> Being able to define templates like this is more a side effect of the new Kconfig preprocessor

Thanks, that was a helpful reference to read (e.g. paying attention that in preprocessor function calls, whitespace is significant and apparently the right style is to avoid it (`$(func,arg1,arg2)`).

> Oh... and if you need someone sympathetic to whine to about MISRA-related cargo-culting, I'm your guy.~

Yeah, maybe we can form a dissident faction ;-).

Thanks for a quick consult! Given that, as mentioned above, nobody seems to disagree, let me close this, but it's certainly a useful reference to have around.

closed 2018-10-07 by pfalcon