zephyrproject-rtos/zephyr #10423: log_core.h error on pointer-to-int-cast on 64bit system

b0661, 2018-10-08:
I´m running an unit test on a 64bit system. The module under test has some logging macros included.

The logging macro tries to cast pointers to u32_t values. This does not work on my system.

The cast should follow the compiler´s pointer size.

```
...sanity-out/unit_testing/tests/unit/drivers/pinctrl/test/main.c:712:3: note: in expansion of macro ‘LOG_ERR’
   LOG_ERR("Invalid pin %d of %s on mux free by %s.",
   ^~~~~~~
...include/logging/log_core.h:147:46: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  log_3(_str, (u32_t)(_arg0), (u32_t)(_arg1), (u32_t)(_arg2), _src_level)
                                              ^

```

nordic-krch, 2018-10-08:
arguments in log message are 32bit width so it will not work in 64bit system, they must be casted. You need to compile unit test in 32 bits. We might in future extend the logger to have argument size according to architecture but there are many other issues pending in the queue so it is rather a long term goal.

b0661, 2018-10-08:
@nordic-krch 

Is there a way I can prevent the logger macros to generate C code at all?

nordic-krch, 2018-10-08:
@b0661 unfortunately, no. Macros were intentionally created in that way to always create code even though it's later on removed during compilation.

closed 2020-02-04 by b0661