zephyrproject-rtos/zephyr #10436: Mess with ssize_t, off_t definitions

pfalcon, 2018-10-08:
Working on https://github.com/zephyrproject-rtos/zephyr/pull/10429 , I was getting weird build failures with native_posix port re: ssize_t and off_t being undefined. Some posix tests had it, some didn't.

Turned out, Zephyr defines these types in fs.h, and in weird way: https://github.com/zephyrproject-rtos/zephyr/blob/master/include/fs.h#L10

So, first of all, ssize_t is being defined only for CONFIG_ARCH_POSIX, and it's defined as:

~~~
typedef __SIZE_TYPE__ ssize_t;
~~~

Then `__SIZE_TYPE__` is:

~~~
$ gcc -m32 -dM -E - </dev/null | grep __SIZE_TYPE__
#define __SIZE_TYPE__ unsigned int
~~~

I like that definition. Really, return type of read, write, etc. should be unsigned, so on 32-bit platform, we had access to (almost) 4GB instead of 2GB, etc. Unfortunately, that's not what POSIX says. As the name itself suggests, ssize_t is a signed variant of size_t, so it cannot be unsigned.

These definitions were added in eb0aaca64d017fad306b040b871e3c0d1b8d6c71 .

Questions:

1. To @ramakrishnapallala as the author: why it was done that way?
2. To @aescolar as the ARCH_POSIX maintainer: are you aware that it's done that way (and what to you think) ?

And the main point: these types should not be defined in "fs.h", as they have nothing to do with "file system". They have to do with generic POSIX I/O model. POSIX says they should be defined in sys/types.h : http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/sys_types.h.html .

Consequently, we either should have them in posix/sys/types.h, or if they're used on Zephyr side (ssize_t definitely is), then in zephyr/types.h .

And the reason I submit this as a ticket and not as PR, because @aescolar, I'd need your consult of where do we stand on the isolation of native_posix port from the host's POSIX includes in general, and whether you anticipate issues with the change as described above. Thanks.

aescolar, 2018-10-08:
@pfalcon: yes, I'm aware of it. It isn't specially beautiful though ( https://github.com/zephyrproject-rtos/zephyr/pull/7326#issuecomment-389539773 )

pfalcon, 2018-10-08:
Ok, so looks like that code comes from you, and just was committed by @ramakrishnapallala .

But anyway, the problem I'm facing now isn't with native_posix, it's actually with all the other boards, and is rooted in the fact that #include_next does not work as expected. Specifically:

~~~
--- a/include/posix/sys/types.h
+++ b/include/posix/sys/types.h
@@ -12,7 +12,9 @@ extern "C" {
 #endif
 
 #ifndef CONFIG_ARCH_POSIX
+before
 #include_next <sys/types.h>
+after
 #endif
~~~

Preprocessor output:

~~~
# 1 "/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/posix/sys/types.h" 1
# 15 "/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/posix/sys/types.h"
before
# 1 "/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/posix/sys/types.h" 1
# 17 "/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/posix/sys/types.h" 2
after
~~~

I.e. #include_next within posix/sys/types.h causes posix/sys/types.h itself be included again. I don't know what may be wrong, maybe #include_next doesn't work for pathed include (like sys/types.h)? Continuing to investigate...

pfalcon, 2018-10-08:
Damn, this time it was fixable by:

~~~
--- a/include/posix/time.h
+++ b/include/posix/time.h
@@ -27,8 +27,8 @@ struct timeval {
 
 #include <kernel.h>
 #include <errno.h>
-#include "sys/types.h"
-#include "signal.h"
+#include <sys/types.h>
+#include <signal.h>
~~~

But mess such a mess...

nashif, 2019-08-29:
stale and no updates for almost a year, lowering priority.

maksimmasalski, 2021-05-20:
> minimal libc's sys/types.h even has:
> 
> ```
> #define unsigned signed
> typedef __SIZE_TYPE__ ssize_t;
> #undef unsigned
> ```
> 
> Cool trick.

Probably it came back, because to be compliant with the MISRA Coding guidelines that part of define should be changed. MISRA says "A macro shall not be defined with the same name as a keyword in C90"

aescolar, 2024-07-09:
Agreed during release meeting to close this as duplicate of #75513

kartben, 2024-07-11:
(closing with the right reason to reflect it was closed as a duplicate)

closed 2024-07-11 by kartben