zephyrproject-rtos/zephyr #10439: Logger calls will execute functions even though the LOG_LEVEL is masked

mike-scott, 2018-10-08:
Here's an example:
```
#if LOG_LEVEL >= LOG_LEVEL_DBG
static char *sprint_token(const u8_t *token, u8_t tkl)
{
...
}
#endif
```
Then later I have code which looks like:
```
LOG_DBG("observer '%s' removed", sprint_token(token, tkl));
```

When DBG is =n, I see the following warnings:
```
zephyr/subsys/net/lib/lwm2m_sock/lwm2m_engine.c:550:3: warning: implicit declaration of function 'sprint_token'; did you mean 'sprintf'? [-Wimplicit-function-declaration]
   sprint_token(token, tkl), lwm2m_sprint_ip_addr(addr));
   ^
```

The behavior has changed here from `SYS_LOG` where `SYS_LOG_DBG` ends up as a no-op.

nashif, 2018-10-08:
thats a "feature", we are using IS_ENABLED AFAIK. 
just remove the #if LOG_LEVEL >= LOG_LEVEL_DBG around sprint_token.

We already found countless log statements that never worked because they were ifdefed like that or were never evaluated.

mike-scott, 2018-10-08:
Won't a needless call to sprint_token will be made if logging is enabled?  This doesn't seem like good behavior.

jukkar, 2018-10-18:
The compiler should optimize away the sprint_token() so there should be no issue here. Closing this atm, re-open if you think this is incorrect.

closed 2018-10-18 by jukkar

mike-scott, 2018-10-18:
I'll just submit the compile warning fix for LwM2M as was discussed.