zephyrproject-rtos/zephyr #10444: tests/subsys/logging/log_core fails on few ARM platforms

nniranjhana, 2018-10-09:
The test fails with an _Assertion Error_ on nRF, FRDM and sam_e70 boards.

**Arch:** ARM
**Platforms:** nrf52_pca10040, nrf52840_pca10056, frdm_k64f, sam_e70_xplained.

**Error log on console:**
```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite test_log_list
 ===================================================================
 starting test - test_log_backend_runtime_filtering
 PASS - test_log_backend_runtime_filtering
 ===================================================================
 starting test - test_log_overflow
 PASS - test_log_overflow
 ===================================================================
 starting test - test_log_arguments
 PASS - test_log_arguments
 ===================================================================
 starting test - test_log_panic
 PASS - test_log_panic
 ===================================================================
 starting test - test_log_from_declared_module
 PASS - test_log_from_declared_module
 ===================================================================
 starting test - test_log_strdup_gc

     Assertion failed at ../zephyr/tests/subsys/logging/log_core/src/log_core_test.c:54: put: (log_msg_source_id_get(msg) not equal to exp_id)

 Unexpected source_id
 
 FAIL - test_log_strdup_gc
 ===================================================================
 starting test - test_strdup_trimming

     Assertion failed at ../zephyr/tests/subsys/logging/log_core/src/log_core_test.c:382: strdup_trim_callback: (len not equal to CONFIG_LOG_STRDUP_MAX_STRING)

 Expected trimmed string
 
 FAIL - test_strdup_trimming
 ===================================================================
 ===================================================================

 PROJECT EXECUTION FAILED
```

nashif, 2018-10-15:
@nordic-krch Are you looking at this, seems to break on most ARM architectures...

nashif, 2018-10-15:
To reproduce this in Qemu, export QEMU_BIN_PATH to point to host qemu (assuming you have a very recent version on the host) and run it on mps2_an385

nashif, 2018-10-17:
@nordic-krch any updates?

PR #10666 (2018-10-17): tests: subsys: logging: Fix log_core test
Log_core test was failing due to test structure elements not being
correctly reset before the test.

Fixes #10444.
closed 2018-10-17 by nashif