zephyrproject-rtos/zephyr #10456: It doesn't support big endian architecture

yashi, 2018-10-09:
The current build system doesn't support big endian architecture.  Even with a compiler which generates big-endian object files, the final linking fails if linker script isn't setup for big endian.   I have two patches to fix this.  One is to add a condition in `include/linker/linker-tool-gcc.h` to control `OUTPUT_FORMAT`.  Another is to add linker command line option `-EB`.

yashi, 2018-10-09:
They are #10457 and #10458

endofexclusive, 2018-10-09:
You may also want to add the option `--big-endian` when calling `gen_isr_tables.py`, for example by adding to `CMakeLists.txt`:
```
if(CONFIG_BIG_ENDIAN)
    list(APPEND GEN_ISR_TABLE_EXTRA_ARG --big-endian)
endif()
```

yashi, 2018-10-10:
@endofexclusive How about like this?

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 5ec15dd169..08ea37b81b 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -762,6 +764,7 @@ if(CONFIG_GEN_ISR_TABLES)
     --output-source isr_tables.c
     --kernel $<TARGET_FILE:zephyr_prebuilt>
     --intlist isrList.bin
+    $<$<BOOL:${CONFIG_BIG_ENDIAN}>:--big-endian>
     $<$<BOOL:${CMAKE_VERBOSE_MAKEFILE}>:--debug>
     ${GEN_ISR_TABLE_EXTRA_ARG}
     DEPENDS zephyr_prebuilt
diff --git a/arch/Kconfig b/arch/Kconfig
index d2bcdd1507..a69fb66907 100644
--- a/arch/Kconfig
+++ b/arch/Kconfig
@@ -39,6 +39,7 @@ config RISCV32
 
 config SPARC
        bool "SPARC architecture"
+       select BIG_ENDIAN
 
 config XTENSA
        bool "Xtensa architecture"
@@ -56,6 +57,9 @@ endchoice
 
 menu "General Architecture Options"
 
+config BIG_ENDIAN
+       bool
+
 config HW_STACK_PROTECTION
        bool "Hardware Stack Protection"
        depends on ARCH_HAS_STACK_PROTECTION

```

yashi, 2018-10-11:
It's closed by #10457.  Thank you, guys.

closed 2018-10-11 by yashi