zephyrproject-rtos/zephyr #10463: can: stm32: Update to new API

erwango, 2018-10-09:
A new CAN API is available in stm32cube packages, current zephyr can driver is implemented on top of legacy driver.

Update stm32 can driver to use new stm32cube API.
Before doing so, check new API is available in all stm32cube packages in order not to introduce fragmentation.

lowlander, 2019-01-19:
What is the status of this? I am currently working on CAN for the STM32F4, and am wondering if I could be of help.

lowlander, 2019-01-29:
@erwango ping ?  what new API do you mean ? The current driver uses the HAL, and old driver from me (that I never upstreamed) the lowlevel api only.

erwango, 2019-01-29:
@lowlander,  HAL CAN API has been deprecated and a new HAL API has been introduced.
We're living on the legacy API since it is still delivered and available under a HAL_CAN_LEGACY_MODULE_ENABLED switch that we define.

alexanderwachter, 2019-02-10:
I prepared a PR for this some time ago. I'll check if this could still be applied.
https://github.com/alexanderwachter/zephyr/commit/d7313cd6db3f63f2d3c339f941bfc0c341c511db

alexanderwachter, 2019-07-01:
#17229 Removes the HAL.

closed 2019-07-19 by alexanderwachter