zephyrproject-rtos/zephyr #10472: some stm32 platforms fail to build w/usb

galak, 2018-10-10:
```
./scripts/sanitycheck -p nucleo_l4r5zi -s samples/net/sockets/dumb_http_server/test
```

Will cause the following error:

```
In file included from /home/galak/git/zephyr/include/arch/arm/arch.h:20:0,
                 from /home/galak/git/zephyr/include/arch/cpu.h:15,
                 from /home/galak/git/zephyr/include/kernel_includes.h:34,
                 from /home/galak/git/zephyr/soc/arm/st_stm32/stm32l4/soc.h:31,
                 from /home/galak/git/zephyr/drivers/usb/device/usb_dc_stm32.c:47:
/home/galak/git/zephyr/drivers/usb/device/usb_dc_stm32.c: In function ‘usb_dc_stm32_get_maximum_speed’:
/home/galak/git/zephyr/sanity-out/nucleo_l4r5zi/samples/net/sockets/dumb_http_server/test/zephyr/include/generated/generated_dts_board.h:438:35: error: ‘ST_STM32_OTGFS_50000000_MAXIMUM_SPEED’ undeclared (first use in this function)
 #define CONFIG_USB_MAXIMUM_SPEED  ST_STM32_OTGFS_50000000_MAXIMUM_SPEED
                                   ^
/home/galak/git/zephyr/sanity-out/nucleo_l4r5zi/samples/net/sockets/dumb_http_server/test/zephyr/include/generated/generated_dts_board.h:438:35: note: in definition of macro ‘CONFIG_USB_MAXIMUM_SPEED’
 #define CONFIG_USB_MAXIMUM_SPEED  ST_STM32_OTGFS_50000000_MAXIMUM_SPEED
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/sanity-out/nucleo_l4r5zi/samples/net/sockets/dumb_http_server/test/zephyr/include/generated/generated_dts_board.h:438:35: note: each undeclared identifier is reported only once for each function it appears in
 #define CONFIG_USB_MAXIMUM_SPEED  ST_STM32_OTGFS_50000000_MAXIMUM_SPEED
                                   ^
/home/galak/git/zephyr/sanity-out/nucleo_l4r5zi/samples/net/sockets/dumb_http_server/test/zephyr/include/generated/generated_dts_board.h:438:35: note: in definition of macro ‘CONFIG_USB_MAXIMUM_SPEED’
 #define CONFIG_USB_MAXIMUM_SPEED  ST_STM32_OTGFS_50000000_MAXIMUM_SPEED
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/galak/git/zephyr/drivers/usb/device/usb_dc_stm32.c:282:11: error: ‘USB_OTG_SPEED_HIGH’ undeclared (first use in this function)
   speed = USB_OTG_SPEED_HIGH;
           ^~~~~~~~~~~~~~~~~~
```

ydamigos, 2018-10-10:
Now it fails with a different error:
```
/home/xekarfwtos/projects/zephyr/drivers/usb/device/usb_dc_stm32.c: In function ‘usb_dc_stm32_get_maximum_speed’: 
/home/xekarfwtos/projects/zephyr/drivers/usb/device/usb_dc_stm32.c:282:11: error: ‘USB_OTG_SPEED_HIGH’ undeclared (first use in this function)
   speed = USB_OTG_SPEED_HIGH;
           ^~~~~~~~~~~~~~~~~~
/home/xekarfwtos/projects/zephyr/drivers/usb/device/usb_dc_stm32.c:282:11: note: each undeclared identifier is reported only once for each function it appears in
```

ydamigos, 2018-10-10:
STM32L4 HAL is missing USB_OTG_SPEED_HIGH definition.

ydamigos, 2018-10-10:
I pushed a commit which adds the missing defines at driver level for L4 series.

PR #10486 (2018-10-10): dts: st: f2/l4: Add missing maximum-speed property
Add missing maximum-speed property to usb nodes
for F2 and L4 series.

Fixes: #10472
commit 84737529 (2018-10-10): dts: st: f2/l4: Add missing maximum-speed property

Add missing maximum-speed property to usb nodes
for F2 and L4 series.

Fixes: #10472
closed 2018-10-10 by galak