zephyrproject-rtos/zephyr #10473: tests/kernel/mem_protect/mem_protect tests are failing on ARM boards

spoorthik, 2018-10-10:
Execution log:

```
***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
Running test suite memory_protection_test_suite
===================================================================
starting test - test_permission_inheritance
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407050
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_permission_inheritance
===================================================================
starting test - test_mem_domain_valid_access
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407cfc
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_mem_domain_valid_access
===================================================================
starting test - test_mem_domain_invalid_access
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407d88
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_mem_domain_invalid_access
===================================================================
starting test - test_mem_domain_partitions_user_rw
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407f40
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_mem_domain_partitions_user_rw
===================================================================
starting test - test_mem_domain_partitions_supervisor_rw
PASS - test_mem_domain_partitions_supervisor_rw
===================================================================
starting test - test_mem_domain_partitions_user_ro
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407f40
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_mem_domain_partitions_user_ro
===================================================================
starting test - test_mem_domain_add_partitions_invalid
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407f40
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_mem_domain_add_partitions_invalid
===================================================================
starting test - test_mem_domain_add_partitions_simple
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20407f40
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
[... 193 lines cut ...]
===================================================================
starting test - test_create_new_higher_prio_thread_from_user
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x204073e4
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_create_new_higher_prio_thread_from_user
===================================================================
starting test - test_create_new_invalid_prio_thread_from_user
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x204074fc
Faulting instruction address = 0x4082cc
Caught system error -- reason 0
FAIL - test_create_new_invalid_prio_thread_from_user
===================================================================
===================================================================

PROJECT EXECUTION FAILED
```

Platforms tested: sam_e70_xplained, nrf52_pca10040, frdm_k64f, nrf52840_pca10056

spoorthik, 2018-10-12:
The test suite fails on freedom_k64f with the fix.

Execution log:

```
***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
Running test suite memory_protection_test_suite
===================================================================
starting test - test_permission_inheritance
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20006884
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_permission_inheritance
===================================================================
starting test - test_mem_domain_valid_access
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x2000753c
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_mem_domain_valid_access
===================================================================
starting test - test_mem_domain_invalid_access
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200075c8
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_mem_domain_invalid_access
===================================================================
starting test - test_mem_domain_partitions_user_rw
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20007780
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_mem_domain_partitions_user_rw
===================================================================
starting test - test_mem_domain_partitions_supervisor_rw
PASS - test_mem_domain_partitions_supervisor_rw
===================================================================
starting test - test_mem_domain_partitions_user_ro
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20007780
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_mem_domain_partitions_user_ro
===================================================================
starting test - test_mem_domain_add_partitions_invalid
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20007780
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_mem_domain_add_partitions_invalid
===================================================================
starting test - test_mem_domain_add_partitions_simple
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20007780
[... 204 lines cut ...]
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_create_new_essential_thread_from_user
===================================================================
starting test - test_create_new_higher_prio_thread_from_user
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20006c18
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_create_new_higher_prio_thread_from_user
===================================================================
starting test - test_create_new_invalid_prio_thread_from_user
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20006d30
Faulting instruction address = 0x96a4
Caught system error -- reason 0
FAIL - test_create_new_invalid_prio_thread_from_user
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

dcpleung, 2018-10-15:
@AdithyaBaglody said that #10542 should fix this.

PR #10696 (2018-10-19): linker: warn about orphan sections (again)
(Previous patch set was reverted due to issue with priv_stack.
 Resubmitting after fixing the faults caused by priv_stack.noinit
 not at the end of RAM.)

This adds a linker flag and necessary changes to linker scripts
so that linker will warn about orphan sections.

Relates to #5534.

Fixes #10473, #10474, #10515.
commit 1fa8cf92 (2018-10-19): linker: arm: place priv.noinit at the end of RAM

This puts the priviledged stack at the end of RAM.
This combines PR #10507 and #10542.

Fixes #10473
Fixes #10474
Fixes #10515
commit 6600c643 (2018-10-19): linker: warn about orphan sections

(Previous patch set was reverted due to issue with priv_stack.
 Resubmitting after fixing the faults caused by priv_stack.noinit
 not at the end of RAM.)

This adds a linker flag and necessary changes to linker scripts
so that linker will warn about orphan sections.

Relates to #5534.

Fixes #10473, #10474, #10515.
closed 2018-10-19 by nashif