zephyrproject-rtos/zephyr #10474: tests/kernel/pipe/pipe test cases are failing on ARM boards

spoorthik, 2018-10-10:
Execution log:
```

***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
Running test suite test_pipe
===================================================================
starting test - test_pipe_on_single_elements
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_on_single_elements
===================================================================
starting test - test_pipe_on_multiple_elements
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_on_multiple_elements
===================================================================
starting test - test_pipe_forever_wait
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_forever_wait
===================================================================
starting test - test_pipe_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_timeout
===================================================================
starting test - test_pipe_get_on_empty_pipe
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_on_empty_pipe
===================================================================
starting test - test_pipe_forever_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_forever_timeout
===================================================================
starting test - test_pipe_get_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_timeout
===================================================================
starting test - test_pipe_get_invalid_size
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x200021d0
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_invalid_size
===================================================================
PROJECT EXECUTION FAILED
```

Platforms tested: frdm_k64f, nrf52840_pca10056, nrf52_pca10040

spoorthik, 2018-10-12:
@nashif @AdithyaBaglody : The test cases still fail even after the fix is merged. 

Execution log:

```
***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
Running test suite test_pipe
===================================================================
starting test - test_pipe_on_single_elements
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_on_single_elements
===================================================================
starting test - test_pipe_on_multiple_elements
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_on_multiple_elements
===================================================================
starting test - test_pipe_forever_wait
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_forever_wait
===================================================================
starting test - test_pipe_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_timeout
===================================================================
starting test - test_pipe_get_on_empty_pipe
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_on_empty_pipe
===================================================================
starting test - test_pipe_forever_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_forever_timeout
===================================================================
starting test - test_pipe_get_timeout
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_timeout
===================================================================
starting test - test_pipe_get_invalid_size
***** USAGE FAULT *****
  Attempt to execute undefined instruction
***** Hardware exception *****
Current thread ID = 0x20002210
Faulting instruction address = 0x7e92
Caught system error -- reason 0
FAIL - test_pipe_get_invalid_size
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

nashif, 2018-10-14:
@dcpleung this is related to the linker changes, we might want to revert them until we have a solution. This turns out to be very fragile.

AdithyaBaglody, 2018-10-15:
@nashif @dcpleung  #10542 Fixes all issues introduced by orphan sections.

commit 1fa8cf92 (2018-10-19): linker: arm: place priv.noinit at the end of RAM

This puts the priviledged stack at the end of RAM.
This combines PR #10507 and #10542.

Fixes #10473
Fixes #10474
Fixes #10515
closed 2018-10-19 by nashif