zephyrproject-rtos/zephyr #10493: native_posix: Warnings on orphan sections after #10368

pfalcon, 2018-10-10:
Resubmitted from https://github.com/zephyrproject-rtos/zephyr/pull/10368#issuecomment-428264786 :

After #10368 was applied, building for BOARD=native_posix produces warnings like below. The resulting executable seems to work.

~~~
[100%] Linking C executable zephyr.elf
/usr/bin/ld: warning: orphan section `.rel._k_mutex.static.pthread_key_lock' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../i386-linux-gnu/Scrt1.o' being placed in section `.rel.dyn'.
/usr/bin/ld: warning: orphan section `.rel._k_mem_slab.static.posix_timer_slab' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../i386-linux-gnu/Scrt1.o' being placed in section `.rel.dyn'.
/usr/bin/ld: warning: orphan section `.rel._k_sem.static.pthread_key_sem' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../i386-linux-gnu/Scrt1.o' being placed in section `.rel.dyn'.
/usr/bin/ld: warning: orphan section `.rel._k_sem.static.mq_sem' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../i386-linux-gnu/Scrt1.o' being placed in section `.rel.dyn'.
/usr/bin/ld: warning: orphan section `.rel.log_const_kernel' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../i386-linux-gnu/Scrt1.o' being placed in section `.rel.dyn'.
Generating files from zephyr.elf for board: native_posix
~~~

pfalcon, 2018-10-10:
cc: @aescolar

aescolar, 2018-10-11:
Raising the priority to medium as this breaks sanitycheck in native_posix fully for "newer" OSs default installations.

dcpleung, 2018-10-11:
GCC on Ubuntu has PIE turned on by default (since 16.10, IIRC). If you do gcc --v and there is an "--enable-default-pie", native_posix will fail to build. Adding "-no-pie" as linker option gets past the PIE issue, but it complains about not able to find "-lgcc_s" (libgcc_s.so and libgcc_s.so.1 are installed by GCC).

@aescolar Are you hitting the PIE issue? If so, we need to create another issue for that.

aescolar, 2018-10-11:
> native_posix will fail to build

With "fail to build" you mean you will get a warning from the linker due to orphan sections?

Sidetrack: I did not try to hint about the PIE issue, but it is also an issue *1. It seems a bit dangerous to try to have this feature on by default in native_posix given that it is built with whatever the user wants/has locally.. 

1* `/usr/bin/x86_64-linux-gnu-ld: warning: orphan section `.rel.log_const_kernel' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib32/Scrt1.o' being placed in section `.rel.dyn'.`

> complains about not able to find "-lgcc_s" (libgcc_s.so and libgcc_s.so.1 are installed by GCC).

I did not see this one, even if I disable PIE with the compile+link options. But I guess I may have just installed more dependencies.

dcpleung, 2018-10-11:
Not due to orphan sections, but something like this:

```
/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: ../libapp.a(main.c.obj): warning: relocation in read-only section `.text.zephyr_app_main'
/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: warning: creating a DT_TEXTREL in object.
```

This causes sanitycheck to fail.

But PR #10528 will fix the orphan section warning emitted.

aescolar, 2018-10-11:
Still some missing ones:
`
/usr/bin/x86_64-linux-gnu-ld: warning: orphan section `.rel.shell_cmd_' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib32/Scrt1.o' being placed in section `.rel.dyn'.
/usr/bin/x86_64-linux-gnu-ld: warning: orphan section `.rel.shell_module_' from `/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib32/Scrt1.o' being placed in section `.rel.dyn'.
`

dcpleung, 2018-10-11:
Hm... I haven't seen the shell ones on my development machine. It was on GCC 6.4.0 though.

The new PR #10536 should fix the shell_* section.

PR #10528 (2018-10-11): linker: more .rel for orphan sections
There are sections defined in common-rom/ram.ld where they are
showing up as orphan sections. So add these as known sections.

Fixes #10493
commit 6ba4488c (2018-10-11): linker: more .rel for orphan sections

There are sections defined in common-rom/ram.ld where they are
showing up as orphan sections. So add these as known sections.

Fixes #10493
PR #10536 (2018-10-12): linker: use wildcards in rel-sections.ld
Update rel-sections.ld to use wildcards instead of
spelling out those sections one by one.

Fixes #10493
commit 9ae3ea47 (2018-10-12): linker: use wildcards in rel-sections.ld

Update rel-sections.ld to use wildcards instead of
spelling out those sections one by one.

Also, for POSIX, don't include this and turns off
the warnings. With different host toolchain across
different OS, it would be maintanence nightmare
to account for all those combinations. So this reverts
the POSIX linker script to before the first orphan
section changes.

Fixes #10493
closed 2018-10-12 by nashif