zephyrproject-rtos/zephyr #10494: sockets: Implement MSG_TRUNC flag for recv()

pfalcon, 2018-10-10:
Feature request from @mike-scott. He'd like to estimate a buffer size required to receive a complete UDP datagram. Per POSIX semantics, a recv() for SOCK_DGRAM will drop any unread data belonging to datagram.

Doing "man recv" on Linux, one can see:

~~~
       MSG_TRUNC (since Linux 2.2)
              For    raw   (AF_PACKET),   Internet   datagram   (since   Linux
              2.4.27/2.6.8), netlink (since Linux 2.6.22), and  UNIX  datagram
              (since  Linux 3.4) sockets: return the real length of the packet
              or datagram, even when it was longer than the passed buffer.
~~~

The expectation is that combined with MSG_PEEK (which we already implement), this flag would give the effect desired by @mike-scott.

TODO:

* Confirm that MSG_TRUNC indeed can be combined with MSG_PEEK
* Analyze POSIX compatibility of this feature

pfalcon, 2018-10-10:
@rlubos : FYI, maybe you'd have comments as the author of MSG_PEEK.

mike-scott, 2018-10-11:
Thanks for adding this issue @pfalcon

rlubos, 2018-10-11:
@pfalcon It was actually some other guy from Nordic, but thanks for letting me know. Good that you've posted the docs for this option, as for instance "man recv" on my Arch doesn't event mention it's existence.

pfalcon, 2018-10-11:
@rlubos: Oops, sorry, I usually check such stuff, but was almost sure it was you ;-).

> for instance "man recv" on my Arch doesn't event mention it's existence.

Hmm, it's not POSIX, but Linux extension, but indeed, a bit strange.

pfalcon, 2018-10-11:
References for recv(..., MSG_PEEK | MSG_TRUNC) used together:

* https://yakking.branchable.com/posts/msg-trunc/
* https://lists.samba.org/archive/samba-technical/2016-February/111982.html

So, I guess that's enough to prove that it's a known pattern (Linux-specific, again).

pfalcon, 2018-10-11:
Now about POSIX (non)compliance:

[recv() spec](http://pubs.opengroup.org/onlinepubs/9699919799/functions/recv.html) doesn't mention anything about MSG_TRUNC, so as was already hinted, it's a Linux extension. That's fine by us, we already implement e.g. MSG_DONTWAIT extension flag.

However, it's interesting to trace this further. MSG_TRUNC actually **is** defined by POSIX, but as *output* flag for [recvmsg()](http://pubs.opengroup.org/onlinepubs/9699919799/functions/recvmsg.html). And see how cunningly it's worded:

> If a message is too long to fit in the supplied buffers, and MSG_PEEK is not set in the flags argument, the excess bytes shall be discarded, and MSG_TRUNC shall be set in the msg_flags member of the msghdr structure.

Note "if ... MSG_PEEK is **not** set". I.e., MSG_TRUNC is a post-factum flag, like if you get data truncation with the buffer, you'll be notified. But reverse task, and subject of the original usecase above - "how to find size of buffer which avoids truncation" isn't seem to be resolvable in bounds of POSIX. Well, the more reasons to implement the Linux extension.

PR #33570 (2021-03-23): net: sockets: Implement MSG_TRUNC flag
Add implementation of the MSG_TRUNC `recv()` flag for UDP and DTLS sockets along with unit tests.

Resolves #10494
closed 2021-03-23 by jukkar