zephyrproject-rtos/zephyr #10499: docs: References to "user threads" are confusing

pfalcon, 2018-10-10:
I was reading https://docs.zephyrproject.org/latest/api/io_interfaces.html?highlight=sensor_trigger_set#_CPPv318sensor_trigger_setP6deviceP14sensor_trigger24sensor_trigger_handler_t , and it says:

> This API is not permitted for user threads.

Well, that's confusing. The opposite of "user thread" is "system thread". What's meant there is "userspace thread". Grepping around, we have a bunch of these "user threads" threads in the docs. But we have enough usages of "userspace" too.

Suggestion: replace "user" in that meaning to &grave;userspace&grave; (note the backticks) through the docbase, then add glossary entry for backticks to link to.

pfalcon, 2018-10-10:
If acked, I can help with this too at suitable time (like, ~ release freeze).

cc: @andrewboie, @nashif

pfalcon, 2018-10-10:
Actually, we probably should turn phrase "This API is not permitted for user threads." into RST warning or admonition.

dbkinder, 2018-10-10:
A grep only shows 8 reST files with the phrase "user thread(s)"  and 5 generated html files from Doxygen API comments, so it's not much to change them all to "userspace thread(s)" if that's the mission here

pfalcon, 2018-10-29:
@dbkinder: Thanks for reply, let's backlog this until ~ release freeze time.

And ping to @andrewboie to avoid surprises when a patch will be prepared/submitted.

pfalcon, 2018-11-01:
> A grep only shows 8 reST files with the phrase "user thread(s)"

Ok, so maybe that's true, but besides that, there're gazillion of cases where "user mode" terminology is used, starting with the section title in https://docs.zephyrproject.org/latest/kernel/usermode/usermode.html . "User mode" is of course more specific term than just "user", but still may be confusing.

Again, the nature of my proposal is to replace ambiguous "user" work in that meaning with much more specific "userspace". Of course, that should be done consistently. So, "user mode" becomes "userspace mode".

That's quite substantial change, and would require pre-ack'ing by @andrewboie and @nashif to avoid monkey's work.

andrewboie, 2018-11-28:
"user thread" is a thread that runs in user mode.

> The opposite of "user thread" is "system thread"

It's currently "supervisor thread".

I don't particularly care what gets used but AFAIK "userspace" is used to describe the overall memory protection feature and "user thread" is used to describe threads that run in user mode.

If you care passionately, send a patch. I have no strong attachment as long as everything is consistent.

PR #35797 (2021-05-29): sensor: use function properties to document the context
We now have function properties which can replace the confusing message
in the doxygen docs.

Fixes #10499
commit 3d39f72a (2021-05-29): sensor: use function properties to document the context

We now have function properties which can replace the confusing message
in the doxygen docs.

Fixes #10499
closed 2021-05-29 by galak