zephyrproject-rtos/zephyr #10503: User define USB function & usb_get_device_descriptor()

aurel32, 2018-10-10:
I am trying to define my own USB function, that is an USB device not defined from one of the existing USB classes. In order to fill the `usb_device_description member of struct usb_cfg_data`, I need to call the `usb_get_device_descriptor()` function. However this function is declared in `subsys/usb/usb_descriptor.h` and thus not part of the public API. I wonder if I am doing something wrong, or if the USB code has to be changed somehow. In that later case I can work on a patch.

One work-around is to define the device as composite even if it has a single function. In that case the call to `usb_get_device_descriptor()` is done from the USB composite code

jfischer-phytec-iot, 2018-10-11:
>One work-around is to define the device as composite even if it has a single function. In that case the call to usb_get_device_descriptor() is done from the USB composite code

That is the way (or should it be). "Legacy" and composite Initialization should be merged/simplified. It is IMHO better to automatically detect if there is more than one class or function, then the composite support does not have to be switched on explicitly.

finikorg, 2018-10-11:
@jfischer-phytec-iot it make sense to merge those interfaces.

aurel32, 2018-10-11:
> That is the way (or should it be). "Legacy" and composite Initialization should be merged/simplified. It is IMHO better to automatically detect if there is more than one class or function, then the composite support does not have to be switched on explicitly.

Thanks for your feedback. I got a quick look at the code. It looks relatively to merge `usb_composite_init()` and `usb_enable()` / `usb_set_config()` into a single function init function that handle both cases, as there is already a lot of code duplication. Choosing automatically between legacy and composite looks a bit more tricky though, at least at a first glance.

Do you plan to work on that? If not I can try to have a first look over the week-end, at least for the first step.

finikorg, 2018-10-16:
OK, sounds good.

carlescufi, 2019-10-09:
Not applicable anymore

closed 2019-10-09 by carlescufi