zephyrproject-rtos/zephyr #10509: tests/kernel/interrupt is failing on NRFx boards

spoorthik, 2018-10-11:
Execution log:
```
**** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
Running test suite interrupt_feature
===================================================================
starting test - test_nested_isr
Triggering irq : 37
isr0 running !!
isr0 execution completed !!

    Assertion failed at zephyr.git/tests/kernel/interrupt/src/nested_irq.c:46: isr0: (new_val not equal to old_val)
Nested interrupt is not working

Triggering irq : 36
isr1 ran !!
FAIL - test_nested_isr
===================================================================
starting test - test_prevent_interruption
PASS - test_prevent_interruption
===================================================================
===================================================================
PROJECT EXECUTION FAILED
```

Platforms tested: nrf52_pca10040, nrf52840_pca10056

nashif, 2018-10-20:
@kumarvikash1 please make sure @ajaykish sees that

kumarvikash1, 2018-10-22:
@nashif , i have informed @ajaykish . I am not sure, why issues are not getting assigned to him.

ajaykish, 2018-10-25:
Looks like due to low power 32 KHz timer in nrf board, we are not achieving the required sequence.
Expected log from this test:
 Triggering irq : 84
 isr0 running !!
 Triggering irq : 83
 isr1 ran !!
 isr0 execution completed !!
 PASS - test_nested_isr
 
 But in nrf board, once isr0 (low priority interrupt) starts execution with busy wait,
 isr1 (high priority interrupt) is not getting triggered meanwhile. 
 Same test is passing with other platforms.

@carlescufi could you please comment on this issue.

PR #11981 (2018-12-11): tests: nested_irq: Fix k_busy_wait usage and interrupt priority
k_busy_wait() call used in test expects time in us, but the test is
specifying wait in ms.

Also the test fails on NRF5 platform as the test hardcodes the interrupts
priority to 0 and 1 and assumes system timer to be of priority 0 which
is not the case in NRF5 platforms as per @pizi-nordic where system timer
is at priority 1. Hence changing test interrupts to 1 and 2.

Fixes: #10509
closed 2018-12-11 by nashif