zephyrproject-rtos/zephyr #10512: Console, logger, shell architecure

nordic-krch, 2018-10-11:
Relations between consoles, logger and shell must be clarified as it generates a lot of confusion as it is now.

Topics to clarify:
- what is console?
- Is it a 'standard io' and thus should be single instance or it's just an 'io' but there is one 'default' (the console).
- What's the purpose of console? Any other use cases than logging and shell?

Current issues with console:
- byte API - slow.  As logger backend candidate throughput must be significantly improved.
- multiple consoles can be enabled but actually one will be used (`__stdout_hook_install`)
- as console is a relatively low level module it is strange to see references to `mcumngr` in `uart_console.c`

It is somehow related to #10480. User expects that enabling shell commands in a module will lead to shell being enabled but as shell is multi-instance it should enable shell with console backend.

nordic-krch, 2018-10-11:
@pfalcon @nashif @carlescufi @jarz-nordic @pizi-nordic @jukkar @aescolar could you comment?
I feel like we should agree on that as way forward is not settled.
@pfalcon you are working around console, maybe you already have this things somehow figured, could you share?

aescolar, 2018-10-11:
One comment here: there is 2 quite different "consoles" in Zephyr.
`drivers/console` which was a set of backends, used by printk and the old shell.
and `subsys/console` (`CONSOLE_SUBSYS`) which was not used inside zephyr itself, but AFAIK pfalcon was promoting as the API to have.

pfalcon, 2018-10-11:
> One comment here: there is 2 quite different "consoles" in Zephyr.

That's right. Here's a short intro into the situation:

The name of "drivers/console" would suggest that those should be *drivers*, and of course the client applications should not call into *drivers*. Instead, apps would go thru "subsystem". That's the pattern we follow in the rest of Zephyr, even if many subsystems are as simple as multiplexors from `struct device` to virtual methods define by a driver for a device.

But drivers/console were special - the apps actually called into particular drivers, and to add support of new driver, one had to patch an application!

That's not the biggest problem with drivers/console though - the API there is rather peculiar. It's line-based, and line editing is handled by the driver! Yes, UART driver handles all those left/right arrow key escape sequences, somewhere straight in the ISR. Needless to say, that line-editing doesn't work for other "console" "drivers" (unless they duplicate code). Another peculiarity was that API was fifo-based, where an app must manage and setup all the buffers for input lines.

There were some other issues, like problems with interrupt handling/buffering, causing character loss on paste from host.

pfalcon, 2018-10-11:
So, to port existing applications to Zephyr, I had a requirement to provide ANSI C/POSIX style functions:

* gets()-alike, to call it and get a line from console
* getchar()-like, to call and get a single char

That's how subsys/console was started. Yes, in a sense, it's "userspace" API, because Zephyr internal components were (and remain) too "originally-designed" to use simple APIs like those.

To implement gets-alike (sane one, called getline()), I was able to base on the API of existing drivers. Implementation of getchar() required working around implementations of the (UART) driver, which soon after was broken. I had little choice as to stop using drivers/console for getchar() at all, and write own "driver" for UART, which so far resides in subsys/console.

aescolar, 2018-10-21:
My 2 cents:

* It would have been nice if the shell & logger would have used the console, as they would have inherited the already available backends. That would have saved work for others, but I can understand they did not fit. From my point of view, no point discussing this.

* About the shell: AFAIK, it seems to assume you always want the embedded Zephyr device to do the shell line editing, and all the fancy stuff which comes with it. If you have a raw, compatible terminal to interact with, and enough RAM and cycles in the embedded device, that works. But, it would seem pretty problematic in other cases. Say, where due to transport latency you want to let the remote terminal handle the line editing, and you only want to transport full lines to the shell. Or say, when you cannot just move the cursor back in the output device.
Meaning, the impression I got is that the shell design is very tailored to a VT100 compatible terminal over UART.

I, personally, do not mind how the shell/logger are architected while they do easily support different backends.
But when it was not working for native_posix, I just got the impression that for the shell, it was way easier to emulate a UART plugged to a raw terminal than to try to get another backend for it working.

jarz-nordic, 2018-10-22:
@aescolar : Thank you for this input. I will work to improve shell according to your suggestions but it will take time... My priority is to fix bug reported by Anas: #10413

pfalcon, 2018-10-31:
So, to not push deds behind the talk (because indeed, we're talking long enough on that, it started long before this ticket), https://github.com/zephyrproject-rtos/zephyr/pull/10765 was the next step.

Capturing @nordic-krch's comments to it:

> as console will be logger/shell backend it should be possible to reconfigure it to synchronous mode in case of panic when data is flushed. In that mode console must not use interrupts or kernel primitives. I only see use case for writing data so it may be that write_sync() would be sufficient. Another option would be to add dedicated function for re-enabling in blocking mode.

Ack from my side on the requirement, we can think what's the cleanest (re: POSIX integration too) and the most general way to implement it.

nordic-krch, 2018-11-02:
My comment: https://github.com/zephyrproject-rtos/zephyr/pull/10765#issuecomment-435284859 is also relevant to that discussion. It's about handling receiver errors on api like the one in #10765:

> How do you see reporting, of RX buffer overflow? It is not hw error but rather application not reading data on time. In that case tty has valid data buffered but tty_read returns error?

pkral78, 2018-11-02:
Some notes:
#10247 was good start, but we should also add support for pooled (non-isr) uart drivers. For example RTT console driver is one that must be implemented by that way. And in fact  #10709 implemented it already for shell subsystem - which is unnecessary effort as my POV is, that shell subsystem could uses console subsystem by default.

Also it would be useful if console subsystem supports multiple ttys on multiple uarts (outside default on for stdin/stdout).

pfalcon, 2018-11-05:
@nordic-krch 

> as console will be logger/shell backend it should be possible to reconfigure it to synchronous mode in case of panic when data is flushed. In that mode console must not use interrupts or kernel [...]

Ok, to support this synchronous/busy-polling mode, there definitely will have to be branch in the code to use either interrupt-driven or busy-polling mode. An condition on which to branch is the size of the corresponding (rx/tx) buffer: indeed, if it's 0, i.e. there's no buffer space, then we can't perform interrupt driven operation.

To make that flexible and configurable, let's just allow to config rx/tx buffer at any time with calls `int tty_set_rx_buf(u8_t *buf, u32_t bufize)`, `int tty_set_tx_buf(u8_t *buf, u32_t bufize)`. Calling these with with `bufsize` of 0 will achieve the desired mode of operation. Note that while original requirement was talking only about write (tx) side of the operation, I'm making this support consistent for both tx and rx.

@nordic-krch, does this sound good, or do you have other ideas/corrections?

pfalcon, 2018-11-05:
@nordic-krch 

> How do you see reporting, of RX buffer overflow? It is not hw error but rather application not reading data on time. In that case tty has valid data buffered but tty_read returns error?

Let's first see how it's done now: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/console/tty.c#L58

So, the original usecase for implementing "buffered UART" (aka "tty") support was:

1. Console, i.e. a device for user interaction. For this usecase, RX buffer overflow is not a problem - there's always a user on the other, who will see that some input they types didn't get thru, and well, will re-type it. The best thing we can do is give some early feedback that overflow happens, so they for example stopped typing too fast (or more realistically, didn't paste so much data at once from clipboard). Indeed, that's what the code linked above would do. Canonically, such a feedback would be a BELL char (\x07), but now it's commonly silenced to avoid physical-level DoS attacks. Indeed, that's how my own terminal emulator is set up. So, instead I decided to output "~", as one of less commonly used chars. As that alone didn't seem as distinguishable enough, I then decided to stuff extra newline "just to be sure". That's not really good, as effectively an amplification - we started with RX overflow, but if for each input overflow char we produce 2 output, we'll end up with output overflow.

But now another usecase is being pushed (by nobody else but me):

2. This same "buffered UART" mechanism is well suited for communicating with UART-interface devices too. Of course "RX overflow feedback" is not just useless, but harmful for this usecase, so we need to: a) either remove it completely; b) make it runtime-configurable based on "console" vs "non-console" usage. And unlike console, RX overflow is fatal-like error in this case. When dealing with device communication, a common usecase we should keep in mind is "length prefix + following data" module. If anything is lost in this case, communication channel is effectively hosed (so this session must be aborted, hopefully using break signalling, if not, by timeout, and new session open).

With these usecases in mind, I'd offer following ways to address them both:

1. Study what POSIX or specific POSIX implementations do in this case - we always should do this.
2. Taking into account that p.1 wasn't yet done, I may theorize on following "original design" solution: a) add concept of "tty mode/flags". Add flag TTY_ERROR_ON_RX_OVERFLOW. If it's set, and RX overflow occurs, next read() will immediately return a suitable error. (Such simplification is made due to concerns expressed above - RX overflow almost certainly means the entire session is hosed, then it's not important if let app read data in buffer and *then* signal overflow error, or signal overflow error right away).

Comments welcome. Note that a similar thought process is what, without a doubt, led POSIX (well, original Unix) to all the termios mess, cooked/raw/semi-cooked/semi-raw/whatever modes, etc. So, we definitely should study prior POSIX art, and better implement its own standard mess, than invent our own original mess ;-).

pfalcon, 2018-11-06:
I wanted to link to this issue: https://github.com/zephyrproject-rtos/zephyr/pull/6815 , where some related ideas on console, etc. were previously discussed.

pfalcon, 2018-11-06:
@pkral78 :

> we should also add support for polled (non-isr) uart drivers.

This is coming, per above requests from @nordic-krch. Even though teh original idea was that "only an interrupt-driven console can be real-world useful". (But certainly, there's no reason to not support even "toy" busy-polling console)

> For example RTT console driver is one that must be implemented by that way. 

It should be noted that using "console" with other devices than "uart" is still an open topic. It's definitely should be supported, but the only detail which is clear now, is that for this, underlying "console driver" should support pull-style API (i.e. getchar/putchar).

> Also it would be useful if console subsystem supports multiple ttys on multiple uarts (outside default on for stdin/stdout).

Yep, this was discussed as one of the possibility. I thought it was in https://github.com/zephyrproject-rtos/zephyr/pull/6815 liked above, but I don't see it there, because previous discussions were spread around various tickets :-(.

For example, we could follow Linux model, where one of the device is "console master", and that's where input comes from, but output may be router to multiple devices. Or maybe we'd bother even with input from multiple devices?

But, before going for anything like that, I'd like to see a real useful usecase behind it (and making a selection between choices like above), not implement it "because we can". As all these features add more and more bloat.

pkral78, 2018-11-06:
> It should be noted that using "console" with other devices than "uart" is still an open topic. It's definitely should be supported, but the only detail which is clear now, is that for this, [...]

Which leads to question, if there should not be some abstraction for both push/pull types of API in between driver and the subsystem. I'm not sure how to call it - stream API, character devices? It can also clarify running context where and when are data handled. I don't see directly hooking console subsystem on uart interrupts is way to go (if we also want to support non-uart drivers). What this part of code does is to store data in ring buffers and synchronize/notify underlying (console) subsystem. Then console subsystem deals with all tty stuff like multiplexing and character handling. Of course that API/layer must be flexible enough as we don't want to copy data between buffers multiple times.

> Yep, this was discussed as one of the possibility. I thought it was in #6815 liked above, but I don't see it there, because previous discussions were spread around various tickets :-(.
>
> For example, [...]

Yeah, but let's start thinking about scenarios like running logger on one rtt channel and shell on second rtt channel or running logger on swo and shell on uart - as both logger and shell may use console subsystem as its IO.

pfalcon, 2018-11-07:
@pkral78:

> Which leads to question, if there should not be some abstraction for both push/pull types of API in between driver and the subsystem.

> I don't see directly hooking console subsystem on uart interrupts is way to go

So, I see the obvious contradiction in what you say. You first ask for supporting push-style API for console, then wonder how UART ISR ended up there. The answer is simple: because UART ISR is a typical example of push-style producer! That's why the *initial*, 1.5yrs old implementation of console looked like that - it was an "impedance matcher" which converted push-style producer(s) into convenient and standard pull-style API. With one example of such producer - UART, and with an initial idea that other producers (e.g. network) can be handled like that too.

It all has changes since then. Console no longer does "impedance matching". Instead, we standardize on pull-style API for "device"/"stream" access. Any "impedance matching" now needs to be done on the subsystem/driver level for a particular device. Console subsystem is nothing more than a multiplexor from "console device" to "concrete stream device(s)", and now we support only "multiplexing" to single device, and that device happens to be only UART (well, "tty").

All that because console subsystem is still in its infancy. You may see that this ticket has "TSC" label, it means that it was/is/to be discussed by Zephyr TSC. And looking at the last TSC notes, the exact question which was posed there by @carlescufi was "Do we need the intermediate console subsystem or not?"

That's the situation we have - some people say: "Hey, we need the console to output to multiple underlying devices!", other people say: "Hey, do we need console at all?"

We'll proceed further only when we get everyone on the same line. And my idea of the same line is simple: "POSIX needs console, so we need console anyway. And then it only makes sense to embrace it as a native Zephyr concept either, instead of trying to hide it in logger, or shell, or somewhere else".

pfalcon, 2018-11-07:
> Of course that API/layer must be flexible enough as we don't want to copy data between buffers multiple times.

We absolutely want to copy data between buffers multiple times. I mean, if the user ordered us that. Just imagine what happens if we want to output character "a" to multiple devices, and one of them is not ready to do that just now. Well, we aren't going to stop the world because of just one lazy device. The solution is to let individual device buffer stuff on their side per their likes (well, per user's likes, as they will configure those devices).

Or maybe we actually want to stop the world. If users asks us to. Like, they say "I have ten devices capable of 1Mbps, and one of 9600bps device. And know what, I want to waste bandwidth of 1Mbps devices. Yeah, all I care is that 9600 device didn't lose a char." And then we decide how to do all that.

So, mult-device support is a ~~swamp~~ task on its own. And IMHO, it's premature to discuss how to support that (in detail) until we decide how to support a mere one-device console and fitted it with the rest of Zephyr subsystems.

pkral78, 2018-11-07:
> So, I see the obvious contradiction in what you say. You first ask for supporting push-style API for console, then wonder how UART ISR ended up there. The answer is simple: because UART ISR is a [...]

This is just misinterpretation of what I pointed on: mixing run context across subsystems, without clear border. This ends up in mess like an old timers API. Another example is USB subsystem, where endpoint enumeration is done directly from isr (at least on stm32) which time to time causes random transfer errors just by enabling higher level of logging. So, because isr is typical push-style producer, lets chain all consumers in isr?

> That's why the _initial_, 1.5yrs old implementation of console looked like that - it was an "impedance matcher" which converted push-style producer(s) into convenient and standard pull-style API. [...]

Nothing wrong with that, but I think that we are talking about an architecture and all I just noted on start of this disccusion is: There should be easy way to integrate RTT with console subsystem.

> It all has changes since then. Console no longer does "impedance matching". Instead, we standardize on pull-style API for "device"/"stream" access. Any "impedance matching" now needs to be done on [...]

Yes, then let's start from real use cases (I described some) and possible solutions in current state and where it should be.

> All that because console subsystem is still in its infancy. You may see that this ticket has "TSC" label, it means that it was/is/to be discussed by Zephyr TSC. And looking at the last TSC notes, [...]

We are aligned with that, but I see that my comments are just too much ff.

pfalcon, 2018-11-07:
> This is just misinterpretation of what I pointed on: mixing run context across subsystems, without clear border.

So, good news is that it's now largely resolved, with more resolutions to gradually perform going forward.

> So, because isr is typical push-style producer, lets chain all consumers in isr?

No, as explained above, we wave push producers goodbye (not composable enough, don't work across domain boundaries).

> There should be easy way to integrate RTT with console subsystem.

Noted. I'm not familiar with RTT enough to give an immediate answer/suggestions, but definitely let's keep it in scope.

> Yes, then let's start from real use cases (I described some) and possible solutions in current state and where it should be.

Sounds good, please lead on that. On my plate so far much more mundane things so far, so while I'm happy to discuss the general mesh of things, specific steps may need to wait its turn.

> We are aligned with that, but I see that my comments are just too much ff.

No, I appreciate the discussion. It allowed for example to collect in this ticket much of the discussion which was spread around the other tickets, and hopefully lay out mid-term plans. Yet, we'd need to have an ordered process, and next step as I see them:

1. Get response from @nordic-krch on  https://github.com/zephyrproject-rtos/zephyr/issues/10512#issuecomment-435914270. Also, feedback from any other interested parties. Ping @nordic-krch ;-). No big hurry, but don't seem too big matters to decide on, so would be nice to it this week.
2. Likewise, response to https://github.com/zephyrproject-rtos/zephyr/issues/10512#issuecomment-435947515
3. Decide on that fate of "tty" naming. Is it good enough? I didn't receive criticism, which is suspicious.
4. Decide of the fate of what "tty" is at all. Based on the current decisions in https://github.com/zephyrproject-rtos/zephyr/pull/10820, it won't be part of UART drivers (instead, #10820 will be, and tty will be on top of it). Then, it's indeed a (mini)subsystem, and should be brushed up as such.

Feedback on all that is welcome.

pfalcon, 2018-11-12:
Ok, I assume there's no criticism of "tty" naming. So, https://github.com/zephyrproject-rtos/zephyr/pull/11298 splits tty-related functions to header tty.h header.

@nordic-krch: Ping on responses to the proposals above on implementation of features requested by you.

aescolar, 2019-01-23:
This topic does not seem to be needing TSC attention today. So removing the TSC label

closed 2022-08-05 by carlescufi