zephyrproject-rtos/zephyr #10515: tests/benchmarks/timing_info faults on ARM platforms

nniranjhana, 2018-10-11:
The time measurement test passes but runs into couple of _USAGE FAULTS_ and _hardware exceptions_ before that.

**Arch:** ARM
**Platforms:** sam_e70_xplained, nrf52_pca10040, nrf52840_pca10056, frdm_k64f

**Error log on console:**
```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 starting test - Time Measurement
 Timing Results: Clock Frequency: 300 MHz
 Context switch                               :4128 cycles , 13760 ns
 Tick overhead                                :1964 cycles ,  6546 ns
 Thread Creation                              :3858 cycles , 12860 ns
 Thread cancel                                :2810 cycles ,  9366 ns
 Thread abort                                 :7576 cycles , 25253 ns
 Thread Suspend                               :5434 cycles , 18113 ns
 Thread Resume                                :5896 cycles , 19653 ns
 Thread Yield                                 :4898 cycles , 16326 ns
 Thread Sleep                                 :5602 cycles , 18673 ns
 Heap Malloc                                  :3738 cycles , 12460 ns
 Heap Free                                    :3108 cycles , 10360 ns
 Semaphore Take with context switch           :6220 cycles , 20733 ns
 Semaphore Give with context switch           :6672 cycles , 22240 ns
 Semaphore Take without context switch        : 498 cycles ,  1660 ns
 Semaphore Give without context switch        :1050 cycles ,  3500 ns
 Mutex lock                                   :1177 cycles ,  3923 ns
 Mutex unlock                                 :1472 cycles ,  4906 ns
 Message Queue Put with context switch        :7056 cycles , 23520 ns
 Message Queue Put without context switch     :1312 cycles ,  4373 ns
 Message Queue get with context switch        :7518 cycles , 25060 ns
 Message Queue get without context switch     :1370 cycles ,  4566 ns
 MailBox synchronous put                      :7820 cycles , 26066 ns
 MailBox synchronous get                      :7454 cycles , 24846 ns
 MailBox asynchronous put                     :1732 cycles ,  5773 ns
 MailBox get without context switch           :3296 cycles , 10986 ns
 
 ***** USAGE FAULT *****
   Attempt to execute undefined instruction
   
 ***** Hardware exception *****
 Current thread ID = 0x20404050
 Faulting instruction address = 0x407a46
 Fatal fault in thread 0x20404050! Aborting.
 User thread Creation                         :9016 cycles , 30053 ns
 
 ***** USAGE FAULT *****
   Attempt to execute undefined instruction

 ***** Hardware exception *****
 Current thread ID = 0x20404050
 Faulting instruction address = 0x407a46
 Fatal fault in thread 0x20404050! Aborting.
 Validation overhead k object init            :1388 cycles ,  4626 ns
 Validation overhead k object permission      :1171 cycles ,  3903 ns
 Timing Measurement  finished
 
 PASS - main
 ===================================================================
 ===================================================================
 PROJECT EXECUTION SUCCESSFUL
```

MaureenHelm, 2018-10-12:
The test fails only with `prj_userspace.conf`, starting with commit 8ce758a8ff1b21e4ce131c79c13f070089418d32

```
8ce758a8ff1b21e4ce131c79c13f070089418d32 is the first bad commit
commit 8ce758a8ff1b21e4ce131c79c13f070089418d32
Author: Daniel Leung <daniel.leung@intel.com>
Date:   Mon Oct 8 10:43:07 2018 -0700

    linker: warn about orphan sections
    
    This adds a linker flag and necessary changes to linker scripts
    so that linker will warn about orphan sections.
    
    Relates to #5534.
    
    Signed-off-by: Daniel Leung <daniel.leung@intel.com>
```

dcpleung, 2018-10-12:
Maybe this is related to #10542?

commit 1fa8cf92 (2018-10-19): linker: arm: place priv.noinit at the end of RAM

This puts the priviledged stack at the end of RAM.
This combines PR #10507 and #10542.

Fixes #10473
Fixes #10474
Fixes #10515
closed 2018-10-19 by nashif