zephyrproject-rtos/zephyr #10516: Migrate drivers to Device Tree

carlescufi, 2018-10-11:
Note: Sensors are covered in #9962

* adc
- [x] dw
- [x] intel_quark_d2000 
- [x] mcux_adc16
- [x] mcux_adc
- [x] nrfx_adc
- [x] nrfx_saadc
- [x] sam_afec
- [ ] ti_adc108s102

* aio
- [ ] comparator_qmsi

* audio
- [N/A] tlv320dac310x

* bluetooth
- [ ] nrf51
- [x] h4
- [x] h5
- [x] spi
- [x] userchan

* can
- [x] stm32

* clock_control
- [ ] beetle
- [?] mcux_ccm
- [?] mcux_sim
- [ ] nrf5
- [ ] quark_se
- [?] stm32

* console
- [N/A] ipm
- [N/A] native_posix
- [ ] ram
- [ ] rtt
- [N/A] telnet
- [x] uart_console
- [x] uart_mcumgr
- [x] uart_pipe
- [N/A] websocket
- [N/A] xtensa_sim

* counter
- [ ] counter_dtmr_cmsdk_apb
- [ ] counter_qmsi_aon
- [ ] counter_qmsi_aonpt
- [ ] counter_tmr_cmsdk_apb
- [ ] timer_dtmr_cmsdk_apb
- [ ] timer_tmr_cmsdk_apb

* crypto
- [ ] ataes132a
- [N/A] mtls_shim
- [N/A] tc_shim

carlescufi, 2019-12-05:
@galak should we close this issue now?

galak, 2022-07-15:
@carlescufi should we close it now :).  I think so.

henrikbrixandersen, 2022-09-05:
> @carlescufi should we close it now :). I think so.

I agree. This issue does not provide value anymore.

closed 2022-09-19 by carlescufi