zephyrproject-rtos/zephyr #10524: PM_CONTROL_OS doesn't work in combination with certain I2C drivers in a single thread context

michiwend, 2018-10-11:
## Summary
When ```PM_CONTROL_OS``` is enabled the OS enters the lowest configured power mode as soon as the I2C transfer function of some I2C drivers gets called in a single thread context.

## Details
Some I2C driver implementations use a semaphore in their transfer implementations and call ```k_sem_take()``` in combination with ```K_FOREVER``` to block the current thread until an I2C event occurs. Since the corresponding ```k_sem_give()``` is called from an interrupt context only and no other thread has been spawned ```_sys_soc_suspend()``` is called with -1 hence the lowest configured power mode is entered.

In our case the OS would go into deep sleep during boot up as soon as any peripheral driver calls the I2C transfer function during its initialization. Especially since the boot up is always single threaded.

### Example: nRF TWI nrfx driver
https://github.com/zephyrproject-rtos/zephyr/blob/2a103ea67493ebcae5c5d7720f6e1a38a05faff1/drivers/i2c/i2c_nrfx_twi.c#L59

https://github.com/zephyrproject-rtos/zephyr/blob/2a103ea67493ebcae5c5d7720f6e1a38a05faff1/drivers/i2c/i2c_nrfx_twi.c#L90

JoeHut, 2018-10-11:
@ramakrishnapallala FYI

ramakrishnapallala, 2019-02-18:
@pizi-nordic FYI

pizi-nordic, 2019-02-19:
This looks like lack of device-level power management.

nashif, 2019-04-02:
i2c and other drivers might require multithreading. this is documented now in the single thread configuration option.

closed 2019-04-02 by nashif