zephyrproject-rtos/zephyr #10531: Default idle thread stack sizes too small when OS controlled power management is used

michiwend, 2018-10-11:
Since ```_sys_soc_suspend()``` is called from the idle thread the default idle thread stack sizes need to be changed to a higher values when ```PM_CONTROL_OS``` is set otherwise a stack overflow can happen.

Tested with the [power_mgr](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/power/power_mgr) sample on an nRF nrf52840_pca10056 devboard (ARM with FPU => 320 idle thread stack size). When stack sentinel is enabled a stack check failure is thrown.

nniranjhana, 2018-10-11:
@ramakrishnapallala FYI..

nashif, 2018-10-16:
@michiwend you can always increase that using Kconfig.

JoeHut, 2018-10-18:
@nashif Sure, but if the power management is done by the OS and a sample is provided it should be ensured that the stack is not corrupted, right?

ramakrishnapallala, 2019-02-18:
@pizi-nordic can you share your inputs?

pizi-nordic, 2019-02-19:
I successfully run the example with stack sentinel and MPU stack protection with idle stack reduced to 288 bytes.

nashif, 2019-02-19:
can't reproduce.

closed 2019-02-19 by nashif