zephyrproject-rtos/zephyr #10535: Failure on tests/posix/common/ due to uninitialized memory access

aescolar, 2018-10-11:
Randomly, the test crashes with an assert. The cause would seem to be the one detected by valgrind (see log below)

#### Steps to reproduce 

On ae8649c5ec59b483615c0348ebcb8da6e346a1ce
```
cmake ../tests/posix/common/ -DBOARD=native_posix -GNinja
ninja
valgrind zephyr/zephyr.exe
```

#### Log
```
***** Booting Zephyr OS v1.9.0-8074-gae8649c5ec *****
Running test suite posix_apis
===================================================================
starting test - test_posix_pthread_execution
==17094== Thread 2:
==17094== Conditional jump or move depends on uninitialised value(s)
==17094==    at 0x10D74C: zap_pthread_attr_setschedparam (pthread.c:89)
==17094==    by 0x10A97E: test_posix_pthread_execution (pthread.c:249)
==17094==    by 0x110800: run_test_functions (ztest.c:58)
==17094==    by 0x110800: test_cb (ztest.c:195)
==17094==    by 0x10C17E: _thread_entry (thread_entry.c:29)
==17094==    by 0x10CE61: posix_thread_starter (posix_core.c:301)
==17094==    by 0x495D3BC: start_thread (pthread_create.c:463)
==17094==    by 0x4A6EE15: clone (clone.S:108)
==17094== 
==17094== Conditional jump or move depends on uninitialised value(s)
==17094==    at 0x10DDE6: valid_posix_policy (pthread_sched.c:12)
==17094==    by 0x10DDE6: zap_sched_get_priority_min (pthread_sched.c:26)
==17094==    by 0x10D75E: is_posix_prio_valid (pthread.c:42)
==17094==    by 0x10D75E: zap_pthread_attr_setschedparam (pthread.c:90)
==17094==    by 0x10A97E: test_posix_pthread_execution (pthread.c:249)
==17094==    by 0x110800: run_test_functions (ztest.c:58)
==17094==    by 0x110800: test_cb (ztest.c:195)
==17094==    by 0x10C17E: _thread_entry (thread_entry.c:29)
==17094==    by 0x10CE61: posix_thread_starter (posix_core.c:301)
==17094==    by 0x495D3BC: start_thread (pthread_create.c:463)
==17094==    by 0x4A6EE15: clone (clone.S:108)
==17094== 
==17094== Conditional jump or move depends on uninitialised value(s)
==17094==    at 0x10DE07: valid_posix_policy (pthread_sched.c:12)
==17094==    by 0x10DE07: zap_sched_get_priority_max (pthread_sched.c:54)
==17094==    by 0x10D76E: is_posix_prio_valid (pthread.c:43)
==17094==    by 0x10D76E: zap_pthread_attr_setschedparam (pthread.c:90)
==17094==    by 0x10A97E: test_posix_pthread_execution (pthread.c:249)
==17094==    by 0x110800: run_test_functions (ztest.c:58)
==17094==    by 0x110800: test_cb (ztest.c:195)
==17094==    by 0x10C17E: _thread_entry (thread_entry.c:29)
==17094==    by 0x10CE61: posix_thread_starter (posix_core.c:301)
==17094==    by 0x495D3BC: start_thread (pthread_create.c:463)
==17094==    by 0x4A6EE15: clone (clone.S:108)
==17094== 

    Assertion failed at /home/alpi/aescolar-ot/tests/posix/common/src/pthread.c:250: test_posix_pthread_execution: (ret not equal to EINVAL)
uninitialized attr set!
FAIL - test_posix_pthread_execution
[...]
```

aescolar, 2018-10-11:
@nashif : Please reassign to whoever is maintaining this area.

spoorthik, 2018-10-12:
#10508 also reports the same issue. Possibly any one of the issues can be closed.

aescolar, 2018-10-12:
Correct, feel free to close this one.