zephyrproject-rtos/zephyr #10537: nRF52 regression: random resets noted across several boards / use cases

mike-scott, 2018-10-12:
When it happens, there is no abort or log related to the cause of the reset.

Example log:
```
[0697100] [fota/lwm2m] [INF] firmware_block_received_cb: 5%
[0698140] [lib/lwm2m_rd_client] [INF] do_update_reply_cb: Update callback (code:2.4)
[0698150] [lib/lwm2m_rd_client] [INF] do_update_reply_cb: Update Done
***** Booting Zephyr OS 1.13.99 *****
[MCUBOOT] [INF] main: Starting bootloader
[MCUBOOT] [INF] boot_status_source: Image 0: magic=good, copy_done=0x1, image_ok=0x1
[MCUBOOT] [INF] boot_status_source: Scratch: magic=bad, copy_done=0x0, image_ok=0x40
[MCUBOOT] [INF] boot_status_source: Boot source: none
[MCUBOOT] [INF] boot_swap_type: Swap type: none
[MCUBOOT] [INF] main: Bootloader chainload address offset: 0x8000
[MCUBOOT] [INF] main: Jumping to the first image slot
***** Booting Zephyr OS 1.13.99 *****
```

This particular reset happened while downloading firmware.  But this can happen at almost any time it seems like.

Bisection leads me to this commit:
https://github.com/zephyrproject-rtos/zephyr/commit/f0a2f2b42300a2ebfb4aaed3c51650d88f38d38a

mike-scott, 2018-10-12:
Hrm, now I just got a random reboot with a build testing Zephyr prior to this commit.   It's very hard to bisect this as you need to log many devices and run them through different use cases and then match up where they rebooted and see if it was an "expected" reboot.

carlescufi, 2018-10-12:
I will copy a few people to see if any of them have observed the same behavior. @nvlsianpu @joerchan @pizi-nordic @anangl.
BTW @mike-scott does this happen only when using MCUboot?

mike-scott, 2018-10-19:
@carlescufi @cvinayak I found the commit that was causing us so many problems:
https://github.com/zephyrproject-rtos/zephyr/commit/9da6a85c8c995f2751ea1a7c7ad80474fb40063b

I'm not familiar with the connection between the BT driver and the flash driver or why this was changed.  So I might need help with a fix.

cvinayak, 2018-10-20:
@mike-scott changes in https://github.com/zephyrproject-rtos/zephyr/commit/9da6a85c8c995f2751ea1a7c7ad80474fb40063b distinguishes the interval for erase and write operations (as compared to previously used single value of that of the erase operation). By using lower value, write operation now should be more frequent while in a BLE connection. This should not introduce regression.

You could assign FLASH_SLOT_WRITE with FLASH_PAGE_ERASE_MAX_TIME_US to get back old behavior, if you want to give it a shot.

mike-scott, 2018-10-20:
@cvinayak we notice the regression under the following setup:
- 6x nRF52832-based HW connected via BLE IPSP (6lowpan)
- 1 Linux gateway (tested w/ previously stable 4.16.x kernel)

Running an MQTT / HTTP client which transfers an OTA and writes it to flash using dfu flash_img API.  When download starts on all 6 clients at least one will generate a VERIFY error and/or a BUS FAULT.

Once reverted, I ran 90 OTAs error free, on 10 devices connected at once.

cvinayak, 2018-10-20:
@mike-scott as the transfer will be much faster, I am suspecting program/ISR  stack overflows leading to memory corruptions, could you paste the BUS FAULT details here?.

mike-scott, 2018-10-20:
I'm packing for ELC-E travel.  However, I can post the BUS FAULT and more information on Sunday/Monday

cvinayak, 2018-10-20:
@mike-scott ok, lets sync-up f2f at ELCE.

cvinayak, 2018-11-02:
@mike-scott any updates?

galak, 2018-11-13:
@mike-scott where are we on this high priority issue?

mike-scott, 2018-11-13:
@galak @carlescufi @cvinayak retesting on today's master.  Will update shortly.

mike-scott, 2018-11-16:
Setup:
- 6x nRF52832-based devices used for the following tests
- Test outline: perform 9x passes of over-the-air tests for each device (that's a total of 9 x 6 = 56 OTAs for a completed test):
    1) test is run with the bluetooth commit in question reverted
    2) test is run with the bluetooth commit left in the source
- Tested yesterday's master branch @ commit 824d0bd854476cf22888bf942ea71f0c4842cef4

Results:
- Test pass #1 ran and completed without issues noted
- Test pass #2 ran and on the 8th pass the following was noted on 1 of the devices:
```
***** BUS FAULT *****
  Instruction bus error
***** Hardware exception *****
Current thread ID = 0x20003394
Faulting instruction address = 0x60586010
Fatal fault in ISR! Spinning...
```

@cvinayak I might need to re-run the test w/ Logger enabled for Bluetooth.  We run with a the following setting which I forgot to remove:
CONFIG_LOG_DEFAULT_LEVEL=0

mike-scott, 2018-11-16:
I turned on logging for:
CONFIG_SOC_LOG_LEVEL_WRN=y
CONFIG_ARCH_LOG_LEVEL_WRN=y
CONFIG_KERNEL_LOG_LEVEL_WRN=y
CONFIG_FLASH_LOG_LEVEL_WRN=y

Re-ran 9 passes and didn't see any errors.

I'll close this until I see this again reliably with more informatio.

closed 2018-11-16 by mike-scott