zephyrproject-rtos/zephyr #10561: Openthread build and usage example

Craigzyc, 2018-10-14:
Can someone help with building zephyr for the nrf52840 including openthread.
I'm using `cmake -GNinja -DBOARD=nrf52840_pca10056 -DCONF_FILE=overlay-ot.conf ..
` in the zephyr/samples/net/echo_client sample but it seems to build identically if I specify the conf or not. Is there something else I need to specify in order to build with openthread support or is it already in zephyr_prebuilt?

It may be working but I can't get access to the cli or any evidence that its actually running. If its working should I be able to type in commands through the serial connection?

Craigzyc, 2018-10-14:
Ok, Found i was using that wrong. 

Using `cmake -GNinja -DBOARD=nrf52840_pca10056 -DOVERLAY_CONFIG=overlay-ot.conf ..`

I get `warning: NET_L2_OPENTHREAD (defined at subsys/net/l2/openthread/Kconfig:12) was assigned the value
'y' but got the value 'n'. You can check symbol information (including dependencies) in the
'menuconfig' interface (see the Application Development Primer section of the manual), or in the
Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NET_L2_OPENTHREAD.html (which is
updated regularly from the master branch). See the 'Setting configuration values' section of the
Board Porting Guide as well.
`

I get this similar warning for a lot of config values. Any idea what I'm doing wrong?
The compiled hex file is also identical in size so I'm guessing openthread wasn't added unless its part of the prebuilt anyways.

rlubos, 2018-10-15:
Hi,
I've just tried `cmake -GNinja -DBOARD=nrf52840_pca10056 -DOVERLAY_CONFIG=overlay-ot.conf ..` and it worked just fine.

Perhaps you didn't clear you build directory after unsuccessfull `cmake` run?

rlubos, 2018-10-15:
> If its working should I be able to type in commands through the serial connection?

Sorry I missed that part - yes, OT shell is exposed to the user, just connect to you board with a terminal, and you should be able to use it.

To activate OT shell type `select ot`, and then type `cmd help` to list supported OT commands. Note that you need to prefix every OT command with `cmd`, so for example to get OT version, use `cmd version` command.

Craigzyc, 2018-10-15:
Ok, yes that worked. Thank you

closed 2018-10-15 by Craigzyc