zephyrproject-rtos/zephyr #10581: [Coverity CID :188744]Memory - corruptions in /drivers/sensor/lis2dh/lis2dh_trigger.c

mandarcthorat1, 2018-10-15:
Static code scan issues seen in File: /drivers/sensor/lis2dh/lis2dh_trigger.c
 Category: Memory - corruptions
 Function: lis2dh_thread_cb
 Component: Drivers
 CID: 188744
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

galak, 2019-03-07:
@avisconti ping.

avisconti, 2019-03-07:
I think it is solved by  802ad61.
I probably forgot to mention it, but it seems exactly same issue as #10571 and #10593.

mandarcthorat1, 2019-03-22:
@avisconti The issue has been again detected by latest coverity scan. Please take a look again.

avisconti, 2019-03-22:
> @avisconti The issue has been again detected by latest coverity scan. Please take a look again.

What is the CID for the new scan?

mandarcthorat1, 2019-03-22:
It has been detected with same CID as before, i.e., 188744.

avisconti, 2019-03-25:
@mandarcthorat1 
I'm not expert on Coverity, but it seems to me that this is maybe a false positive, I don't think there will be
any memory corruption. Since atomi_test_and_clear_bit() is used in other part of the code as well is it possible to see whether the same issue has been issued already and in thata case how it has been closed? Thanks.

galak, 2019-03-26:
Closing and update coverity as false positive.

closed 2019-03-26 by galak