zephyrproject-rtos/zephyr #10592: [Coverity CID :188731]Memory - illegal accesses in /ext/hal/ti/simplelink/source/ti/drivers/net/wifi/source/driver.c

mandarcthorat1, 2018-10-15:
Static code scan issues seen in File: /ext/hal/ti/simplelink/source/ti/drivers/net/wifi/source/driver.c
 Category: Memory - illegal accesses
 Function: _SlDrvMsgRead
 Component: Networking
 CID: 188731
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

GAnthony, 2018-11-07:
Filed a TI JIRA ticket on the TI host driver: (internal link) https://jira.itg.ti.com/browse/CC3X20SDK-1578 to track.
Should be fixed in a future TI SimpleLink SDK release.

vanti, 2019-02-13:
Here's the analysis from the TI host driver team:

> This code is relevant for multiple select only and actually releases the control socket. The problematic code section will never be reached since LengthToCopy will be always 0. This is due to the fact that only 4 bytes of data are being sent.
> 
> This code should be removed.

Given this is dead code, we should continue to ignore this coverity finding til the team provides an official fix.

galak, 2019-02-13:
Yeah, we don't normally scan ext/ so going to close this issue.

closed 2019-02-13 by galak