zephyrproject-rtos/zephyr #10610: i2s driver for stm still uses old logger

nashif, 2018-10-15:
@avisconti 

commit 7b82e9f2d09ca17a97e34d11965e119f394ee39c
Author: Armando Visconti <armando.visconti@st.com>
Date:   Thu Jun 14 09:27:46 2018 +0200

    drivers: i2s: Add support for I2S stm32

Please move this driver to new logger

avisconti, 2018-10-15:
Sure. I'll do it tomorrow.

avisconti, 2018-10-15:
@nashif 
How to test it? Porting new logging method was easy enough for doing it right now. But tomorrow I would like to see if I can really see the logs.

avisconti, 2018-10-16:
OK, it works. Nevertheless there are too many logs and I didn't get how to filter them. For example I may want just to see logs from my I2S driver, while now I get logs also from dma driver. Not sure if it is possible to filter the driver from menuconfig.

avisconti, 2018-10-24:
@nashif 
You may close this issue

closed 2018-11-05 by nashif