zephyrproject-rtos/zephyr #10611: STM32L4 GPIOC Missing Interrupt support?

sdrsdr, 2018-10-15:
Similar to #9630 when I try to use interrupt callback related to  GPIOC  the callback is not triggered. GPIOA  and GPIOB are working as expected. The code I use for testing is:
```
#include <zephyr.h>
#include <gpio.h>
#include <misc/printk.h>

	struct device *gpioa;
	struct device *gpiob;
	struct device *gpioc;

static struct gpio_callback btn_gpioa_cb;
static struct gpio_callback btn_gpiob_cb;
static struct gpio_callback btn_gpioc_cb;

void _on_button(struct device *gpio, struct gpio_callback *cb,
		u32_t pins)
{
	if (gpio==gpioa) {
		printk("GPIOA pins %08x\r\n",  pins);
	} else if(gpio==gpiob) {
		printk("GPIOB pins %08x\r\n",  pins);
	} else if(gpio==gpioc) {
		printk("GPIOC pins %08x\r\n",  pins);
	} else {
		printk("port?? %p  pins %08x\r\n", gpio,  pins);
	}
}

void _init_buttons(void)
{
	gpioa = device_get_binding("GPIOA");
	gpiob = device_get_binding("GPIOB");
	gpioc = device_get_binding("GPIOC");

	gpio_pin_configure(gpioa, 10,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_pin_configure(gpiob, 4,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_pin_configure(gpiob, 5,
			   (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
			    GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP));

	gpio_init_callback(&btn_gpioa_cb, _on_button, BIT(10));
	gpio_init_callback(&btn_gpiob_cb, _on_button, BIT(4) | BIT(5));
	gpio_init_callback(&btn_gpioc_cb, _on_button, BIT(5));

	if (gpio_add_callback(gpioa, &btn_gpioa_cb) < 0) {
		printk("btn cb failed %s\r\n", "GPIOA");
		return;
	} else {
		printk("GPIOA is %p\r\n", gpioa);
	}

	if (gpio_add_callback(gpiob, &btn_gpiob_cb) < 0) {
		printk("btn cb failed %s\r\n", "GPIOB");
		return;
	} else {
		printk("GPIOB is %p\r\n", gpiob);
	}

	if (gpio_add_callback(gpioc, &btn_gpioc_cb) < 0) {
		printk("btn cb failed %s\r\n", "GPIOC");
		return;
	} else {
		printk("GPIOC is %p\r\n", gpioc);
	}

	gpio_pin_enable_callback(gpioa, 10);
	
	gpio_pin_enable_callback(gpiob, 4);
	gpio_pin_enable_callback(gpiob, 5);
	
	gpio_pin_enable_callback(gpioc, 5);
}

void main(void)
{
	_init_buttons();

	while (true) {
		k_sleep(K_SECONDS(1));
		printk(".");
	}
}
``` 
project file is minimal:
```
CONFIG_GPIO=y
CONFIG_SERIAL=y

CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
```

make menuconfig does not show usart3 selected so gpioc5 should be free for gpio..

erwango, 2018-10-15:
Hi,

Seems you miss the `gpio_init_callback` for `btn_gpioc_cb` in the example you gave.
I had a try on my board and it worked.

Can you have another check?

sdrsdr, 2018-10-15:
gpio_pin_configure is indeed missing :-( I'll check again tomorrow

sdrsdr, 2018-10-16:
Ok here is the revised testing code:
```
#include <zephyr.h>
#include <gpio.h>
#include <misc/printk.h>

        struct device *gpioa;
        struct device *gpiob;
        struct device *gpioc;

static struct gpio_callback btn_gpioa_cb;
static struct gpio_callback btn_gpiob_cb;
static struct gpio_callback btn_gpioc_cb;

void _on_button(struct device *gpio, struct gpio_callback *cb,
                u32_t pins)
{
        if (gpio==gpioa) {
                printk("GPIOA pins %08x\r\n",  pins);
        } else if(gpio==gpiob) {
                printk("GPIOB pins %08x\r\n",  pins);
        } else if(gpio==gpioc) {
                printk("GPIOC pins %08x\r\n",  pins);
        } else {
                printk("port?? %p  pins %08x\r\n", gpio,  pins);
        }
}

void _init_buttons(void)
{
        gpioa = device_get_binding("GPIOA");
        gpiob = device_get_binding("GPIOB");
        gpioc = device_get_binding("GPIOC");

        if (gpio_pin_configure(gpioa, 10,
                           (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
                            GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP))
        ){
                printk("GPIOA PIN 10 config failed!\r\n");
        } else {
                printk("GPIOA PIN 10 config OK\r\n");
        };

        if(gpio_pin_configure(gpiob, 4,
                           (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
                            GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP))
        ){
                printk("GPIOB PIN 4 config failed!\r\n");
        } else {
                printk("GPIOB PIN 4 config OK\r\n");
        };

        if(gpio_pin_configure(gpiob, 5,
                           (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
                            GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP))
        ){
                printk("GPIOB PIN 5 config failed!\r\n");
        } else {
                printk("GPIOB PIN 5 config OK\r\n");
        };

        if(gpio_pin_configure(gpioc, 5,
                           (GPIO_DIR_IN | GPIO_INT | GPIO_INT_EDGE |
                            GPIO_INT_DOUBLE_EDGE | GPIO_PUD_PULL_UP))
        ){
                printk("GPIOC PIN 5 config failed!\r\n");
        } else {
                printk("GPIOC PIN 5 config OK\r\n");
        };

        gpio_init_callback(&btn_gpioa_cb, _on_button, BIT(10));
        gpio_init_callback(&btn_gpiob_cb, _on_button, BIT(4) | BIT(5));
        gpio_init_callback(&btn_gpioc_cb, _on_button, BIT(5));

        if (gpio_add_callback(gpioa, &btn_gpioa_cb) < 0) {
                printk("btn cb failed %s\r\n", "GPIOA");
                return;
        } else {
                printk("GPIOA is %p\r\n", gpioa);
        }

        if (gpio_add_callback(gpiob, &btn_gpiob_cb) < 0) {
                printk("btn cb failed %s\r\n", "GPIOB");
                return;
        } else {
                printk("GPIOB is %p\r\n", gpiob);
        }

        if (gpio_add_callback(gpioc, &btn_gpioc_cb) < 0) {
                printk("btn cb failed %s\r\n", "GPIOC");
                return;
        } else {
                printk("GPIOC is %p\r\n", gpioc);
        }

        gpio_pin_enable_callback(gpioa, 10);

        gpio_pin_enable_callback(gpiob, 4);
        gpio_pin_enable_callback(gpiob, 5);

        gpio_pin_enable_callback(gpioc, 5);
}

void main(void)
{
        _init_buttons();

        while (true) {
                k_sleep(K_SECONDS(1));
                printk(".");
        }
}
```
this time all pins and structs are properly configured and the callback on PC5 is triggered but not on PB5. When I don't configure pin5 on gpioc the callback is triggered again on PB5. There is some remote memory about all ports are sharing interrupt  mask per pin so PC5 and PB5 are the same interrupt and mask and you have to somehow check which pin on which port is triggering the interrupt basically by not intermixing same pins on ports as interrupt sources but if this is the case a error should be returned or both callbacks should be trigered .. or something ..

erwango, 2018-12-04:
@sdrsdr, sorry I've missed your last message. I'll have a look to your point this week. Do you confirm it's still valid?

erwango, 2018-12-04:
@sdrsdr, I had a quick look, and indeed issue seems to be located in the fact we're using EXTI HW as interrupt controller for GPIOs. Depending on STM32 series, EXTI supports more or less configurable lines. I advise you to check Reference Manual DM00083560, chapter 12.4.
To sum up all GPIO_Y.PinX are mapped on LineX. Then we'll have dedicated configuration capability for this line that will be shared for all GPIO_Y.PinX (for instance trigger will be common).
Today, EXTI driver have be implemented on that model and all GPIOX.pinY share the same configuration (trigger and callback).
For callback, as this is pure software, we can enhance support and get one cb per GPIOX.pinY, at expense of footprint though. Though, triggers are HW dependent and this cannot be extended.

Hence, we won't be able to totally have independent configuration for each GPIOX.pinY. I'll have a try and check how simply we can add  callbacks per GPIOX.pinY, but to me this is rather a enhancement than a bug fix as this is HW limitation we're facing here.

erwango, 2018-12-04:
@sdrsdr I had a look in detail and not sure excactly what would be the good option here.
Adding support for callbacks per GPIOX.PinY is possible but then both callbacks GPIOA.PIN5 and GPIOC.PIN5 would be triggered in case one of PIN5 is triggered, but not way to know the GPIO... I'm not sure there is anything we can do with that.

About providing an error message at configuration, there is already an __ASSERT that checks if a callback is already registered on the line. Though, since disabled by default it is a bit useless.
What can be done is to change the ASSERT to a return -EIO.

sdrsdr, 2018-12-06:
Great! :)

commit ea1f2b60 (2018-12-07): drivers/gpio: stm32: return error when pin line already in use

GPIO pin interrupts share common EXTI resources.
Return an error when attempt to configure a line already in use.

Fixes #10611
closed 2018-12-06 by sdrsdr