zephyrproject-rtos/zephyr #10619: [Zephyr] Firmware upgrade through UART [Nordic 52840 - pca 10056 Board]

parinpatel, 2018-10-16:
Hi all,

[NRF52840 - pca10056]

I used the mcuboot for bootloader 
and I wanted to firmware upgrade using UART interface 
I also found that there is one flag "CONFIG_MCUBOOT_SERIAL" which is used for firmware upgrade through UART but after enable this , 
Is there any command line utility or any step which are required for flashing the firmware through UART?? 
Currently I am unable to find any document or flashing steps for it. 
Please help me to solve this issue.

For nordic there is one utility "nrfutil" and its helpful for firmware flash using UART for nordic stack.

parinpatel, 2018-10-16:
Currently I found some command from below link 

https://github.com/zephyrproject-rtos/zephyr/issues/9509

Is it correct?

carlescufi, 2018-10-16:
@parinpatel can you comment on #9509 instead?

parinpatel, 2018-10-16:
Ok

closed 2018-10-16 by parinpatel