zephyrproject-rtos/zephyr #10623: tests/drivers/watchdog/wdt_basic_api fails on Quark platforms

nniranjhana, 2018-10-16:
On quark_se_c1000, #8097 says its fixed, but recent builds show an Assertion Failure, not a test crash like reported before. This seems to be a different issue, with the assertion failure and restarts arising after #9872 was merged.

**Arch:** x86, ARC
**Platforms:** quark_se_c1000, quark_d2000

**Error on console log:**

_quark_se_c1000:arc_

```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Testcase passed
 Testcase: test_wdt_callback_1
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_callback_1
 
     Assertion failed at /../zephyr/tests/drivers/watchdog/wdt_basic_api/src/test_wdt.c:295: test_wdt: test_wdt_callback_1() == TC_PASS is false
 
 FAIL - test_wdt
 ===================================================================
 ===================================================================

 PROJECT EXECUTION FAILED
```

On _quark_d2000:x86_, it restarts multiple times without any terminating output:

```
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU
 ***** delaying boot 1000ms (per build configuration) *****
 ***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
 Running test suite wdt_basic_test
 ===================================================================
 starting test - test_wdt
 Testcase: test_wdt_no_callback
 Waiting to restart MCU

 description: testcase didn't run as expected
```

spoorthik, 2018-10-16:
@nniranjhana: It is duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/8097

nniranjhana, 2018-10-17:
When flashed directly on both the platforms (without infrastructure), the output is the same.

The test passes on x86 core of the quark_se_c1000 platform. If the test should not be running on the arc core, we need to exclude in the .yaml file.

The test restarts multiple times on x86 core of the quark_d2000, same as above.

nashif, 2019-01-30:
duplicate of #8097

closed 2019-01-30 by nashif