zephyrproject-rtos/zephyr #10628: tests/kernel/common crashes on ESP32

spoorthik, 2018-10-16:
Execution log:

```
***** Booting Zephyr OS zephyr-v1.13.0-1017-gc0a6801 (delayed boot 1000ms) *****
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 3 PS 0x00060020 PC 0x400838fe VADDR 0x40080480
 **  A0 0x80083a16  SP 0x3ffb5100  A2 0x00000002  A3 0x40080474
 **  A4 0x00000001  A5 0x00000000  A6 0x3ffb3748  A7 0x00000000
 **  A8 0x800838f4  A9 0x3ffb50e0 A10 0x00098968 A11 0x3ffb51e0
 ** A12 0x00000024 A13 0x00000004 A14 0x3ffb009c A15 0x3ffb006c
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb3e08
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```

Steps to reproduce:
```
cd tests/kernel/common
Add CONFIG_BOOT_DELAY=1000 to prj.conf
mkdir build && cd build
cmake -DBOARD=esp32 .. 
make flash
```

nniranjhana, 2018-10-16:
The tests in posix/fs also crash with similar log on ESP32:

```
***** Booting Zephyr OS 1.13.99 (delayed boot 1000ms) *****
** FATAL EXCEPTION
** CPU 0 EXCCAUSE 3 PS 0x00060020 PC 0x400816d6 VADDR 0x4008048c
** A0 0x800817ee SP 0x3ffb20f0 A2 0x00000002 A3 0x40080480
** A4 0x00000001 A5 0x00000000 A6 0x3ffb1a60 A7 0x00000000
** A8 0x800816cc A9 0x3ffb20d0 A10 0x00098968 A11 0x3ffb21d0
** A12 0x00000024 A13 0x00000004 A14 0x3ffb00a8 A15 0x3ffb0078
** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
** SAR 0x00000000
Current thread ID = 0x3ffb1cf8
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```

henrikbrixandersen, 2019-03-12:
This is likely related to `CONFIG_LOG=y`, similar to #13291.

chrta, 2019-06-18:
As an update, this happens in the current master branch:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.14.0-1292-gca49d6a85776 (delayed boot 1000ms) *****
Running test suite common
===================================================================
starting test - test_verify_bootdelay
PASS - test_verify_bootdelay
===================================================================
starting test - test_irq_offload
PASS - test_irq_offload
===================================================================
starting test - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
starting test - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
starting test - test_atomic
PASS - test_atomic
===================================================================
starting test - test_bitfield
twiddling bits....
PASS - test_bitfield
===================================================================
starting test - test_printk
22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0xcafebabe 0x0000beef
0x1 0x01 0x0001 0x00000001 0x0000000000000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef  0x0000002a      42
ERR -1 ERR ffffffffffffffff
PASS - test_printk
===================================================================
starting test - test_slist
PASS - test_slist
===================================================================
starting test - test_dlist
PASS - test_dlist
===================================================================
starting test - test_intmath
PASS - test_intmath
===================================================================
starting test - test_timeout_order
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 3 PS 0x00060020 PC 0x400845cc VADDR 0x400805a0
 **  A0 0x8008480c  SP 0x3ffb65b0  A2 0x40080594  A3 0x00000001
 **  A4 0x00000000  A5 0x3ffb6b50  A6 0x00000001  A7 0x00000000
 **  A8 0x800845c2  A9 0x3ffb6580 A10 0x00000000 A11 0x00000000
 ** A12 0x00000000 A13 0x3ffb4304 A14 0x3ffb4304 A15 0x3ffb42e0
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb41bc
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
thread 0 synced on timer 0
@@
```

carlescufi, 2020-03-16:
@nashif is this still an issue?

chrta, 2020-03-19:
This seems to be fixed for the current master branch (also tests/posix/fs works) :

```
***** delaying boot 1000ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v2.2.0-480-g495407aabc63  (delayed boot 1000ms) ***
Running test suite common
===================================================================
starting test - test_verify_bootdelay
PASS - test_verify_bootdelay
===================================================================
starting test - test_irq_offload
PASS - test_irq_offload
===================================================================
starting test - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap
===================================================================
starting test - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap
===================================================================
starting test - test_sys_get_be64
PASS - test_sys_get_be64
===================================================================
starting test - test_sys_put_be64
PASS - test_sys_put_be64
===================================================================
starting test - test_sys_get_be32
PASS - test_sys_get_be32
===================================================================
starting test - test_sys_put_be32
PASS - test_sys_put_be32
===================================================================
starting test - test_sys_get_be16
PASS - test_sys_get_be16
===================================================================
starting test - test_sys_put_be16
PASS - test_sys_put_be16
===================================================================
starting test - test_sys_get_le16
PASS - test_sys_get_le16
===================================================================
starting test - test_sys_put_le16
PASS - test_sys_put_le16
===================================================================
starting test - test_sys_get_le32
PASS - test_sys_get_le32
===================================================================
starting test - test_sys_put_le32
PASS - test_sys_put_le32
===================================================================
starting test - test_sys_get_le64
PASS - test_sys_get_le64
===================================================================
starting test - test_sys_put_le64
PASS - test_sys_put_le64
===================================================================
starting test - test_atomic
PASS - test_atomic
===================================================================
starting test - test_bitfield
twiddling bits....
PASS - test_bitfield
===================================================================
starting test - test_printk
22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0xcafebabe 0x0000beef
0x1 0x01 0x0001 0x00000001 0x0000000000000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef  0x0000002a      42
ERR -1 ERR ffffffffffffffff
PASS - test_printk
===================================================================
starting test - test_timeout_order
thread 0 synced on timer 0
thread 1 synced on timer 1
thread 2 synced on timer 2
PASS - test_timeout_order
===================================================================
starting test - test_clock_uptime
PASS - test_clock_uptime
===================================================================
starting test - test_clock_cycle
PASS - test_clock_cycle
===================================================================
starting test - test_version
PASS - test_version
===================================================================
starting test - test_multilib
PASS - test_multilib
===================================================================
starting test - test_thread_context
PASS - test_thread_context
===================================================================
starting test - test_bounds_check_mitigation
PASS - test_bounds_check_mitigation
===================================================================
Test suite common succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

closed 2020-08-31 by nashif