zephyrproject-rtos/zephyr #10632: tests/kernel/poll faults on freedom_k64f

spoorthik, 2018-10-16:
Execution log:

```
***** Booting Zephyr OS zephyr-v1.13.0-1017-gc0a6801 (delayed boot 1000ms) *****
Running test suite poll_api
===================================================================
starting test - test_poll_no_wait
***** BUS FAULT *****
  Stacking error
  NXP MPU error, port 3
    Mode: Supervisor, Data Address: 0x20002d10
    Type: Write, Master: 0, Regions: 0x8200
***** Hardware exception *****
Current thread ID = 0x200039b8
Faulting instruction address = 0x20002b18
Fatal fault in thread 0x200039b8! Aborting.
***** Debug monitor exception (not implemented) *****
***** Hardware exception *****
Current thread ID = 0x20000368
Faulting instruction address = 0x3ca0
Fatal fault in essential thread! Spinning...
```
Steps to reproduce:
```
cd tests/kernel/poll
mkdir build && cd build
cmake -DBOARD=freedom_k64f
make flash
```

dcpleung, 2018-10-16:
Reverting this

```
commit 3fb6ea210a0742f1c223c96d6c7cfb937bd873aa
Author: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
Date:   Tue Oct 9 23:05:13 2018 +0200

    arch: arm: re-program MPU stack guard upon userspace enter
    
    Before entering userspace, the MPU stack guard is
    configured to guard the default (i.e. afterwards, the "user")
    thread stack, as the privileged stack has not yet been set. Upon
    entering userspace the MPU stack guard needs to be re-programmed
    to the privileged stack, otherwise, there is a risk that the
    first system call might be serviced without privileged stack
    protection, if it occurs before any context-switch. This commit
    fixes this issue, by calling configure_mpu_stack_guard(..) upon
    userspace entering.
    
    Signed-off-by: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
```

and the test passed on frdm_k64f and nrf52_pca10040.

dcpleung, 2018-10-17:
The test passed on nrf52_pca10040 after increasing PRIVILEGED_STACK_SIZE to 512.

PR #10729 (2018-10-19): arch: increase rivileged stack with mpu stack guard
After commit 3fb6ea21 the stack is not enough and a few tests have been
failing.

Fixes #10632
commit 005f6c7a (2018-10-19): arch: increase privileged stack with mpu stack guard

After commit 3fb6ea21 the stack is not enough and a few tests have been
failing.

Fixes #10632
closed 2018-10-19 by nashif