zephyrproject-rtos/zephyr #10633: tests/kernel/mem_protect/syscall crashes on freedom_k64f

spoorthik, 2018-10-16:
Execution log:

```
***** Booting Zephyr OS zephyr-v1.13.0-1017-gc0a6801 (delayed boot 1000ms) *****
Running test suite syscalls
===================================================================
starting test - test_string_nlen
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xfffffff0
PASS - test_string_nlen
===================================================================
starting test - test_string_nlen
***** BUS FAULT *****
  Stacking error
  NXP MPU error, port 3
    Mode: Supervisor, Data Address: 0x20001a10
    Type: Write, Master: 0, Regions: 0x8200
***** Hardware exception *****
Current thread ID = 0x20001bc4
Faulting instruction address = 0xaac0460
Fatal fault in thread 0x20001bc4! Aborting.
```

dcpleung, 2018-10-16:
Reverting this

```
commit 3fb6ea210a0742f1c223c96d6c7cfb937bd873aa
Author: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
Date:   Tue Oct 9 23:05:13 2018 +0200

    arch: arm: re-program MPU stack guard upon userspace enter
    
    Before entering userspace, the MPU stack guard is
    configured to guard the default (i.e. afterwards, the "user")
    thread stack, as the privileged stack has not yet been set. Upon
    entering userspace the MPU stack guard needs to be re-programmed
    to the privileged stack, otherwise, there is a risk that the
    first system call might be serviced without privileged stack
    protection, if it occurs before any context-switch. This commit
    fixes this issue, by calling configure_mpu_stack_guard(..) upon
    userspace entering.
    
    Signed-off-by: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
```

and the test passed on frdm_k64f and nrf52_pca10040.

dcpleung, 2018-10-17:
The test passed on nrf52_pca10040 after increasing PRIVILEGED_STACK_SIZE to 512.

ioannisg, 2019-02-05:
I think this was a problem of not sufficient size for supervisor threads.
Anyways, we can close this after we merge #12817; the tests are passing on _both_ frdm_k64f, and nrf52_pca10040. (may be already passing now, but we can wait a bit)